LibreCAD/LibreCAD #2062: It would be handy if file save as populated the current file name rather than Untitled

jpagliaccio, 2025-03-06:
It would be handy if file save as populated the current file name rather than Untitled. It's a common behavior and allows for filename corrections and helps with versioning.

sand1024, 2025-03-06:
hm... not sure I understand this, sorry. 

Current file name instead of Untitled... but what is the current file name if the file was not saved before?

dellus, 2025-03-07:
I think jpagliaccio is talking of "save as" for an existing and altered drawing. If the original drawing is named "drawing.dxf" the input box offers "untitled.dxf" and you have to type in the name completely new. 
If the old name is offered for "save as" there is the danger of accidentally overwriting the existing drawing with the altered one, if you miss to modify the name.
Maybe the the old name could be offered with an addition, like "drawing_1.dxf" or "drawing_" to prevent overwriting and also easing version naming.

jpagliaccio, 2025-03-07:
If the current file is new, without a name yet, then "Untitled" is okay. That is just a save. 

If you are doing a "Save As" with an existing file, it's common and handy to have the existing filename show. So then you could simply make a change like: motor_bracket_v01.dxf to motor_bracket_v02.dxf.  

Following "standard" and common behavior for little things like this makes the app more user friendly.

sand1024, 2025-04-23:
@jpagliaccio 
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

Please check and close if completed.

sand1024, 2026-09-26:
@sand1024 still open, as there is no "Untitled" for newly created file (fail). 
for saving existing one - current file name is used (ok)