LibreCAD/LibreCAD #2064: Crash on editing block on master branch

qwilvove, 2025-03-08:
Hello,

I have noticed that LibreCAD crashes when we attempt to edit a block. This crash is currently affecting the master branch of the project.

Regards

sand1024, 2025-03-08:
Could you please provide more details or, ideally, sample dxf where such issue is reproduced?

qwilvove, 2025-03-09:
For example, when I start LibreCAD with no dxf, I go to blocks widget to create a new empty block, then I give a name to the new block, after that, if I try to edit this new block, LibreCAD crashes.

Seems to be the same issue as this : https://github.com/LibreCAD/LibreCAD/issues/1886

dxli, 2025-03-09:
not related to #1886 , this is a regression from 05bc6ccb

qwilvove, 2025-03-09:
Do I need to close the issue ?

sand1024, 2025-03-09:
@dxli If it's related to Informational Cursor, I think the fix was added as part of PR that was not merged yet

closed 2025-03-23 by dxli