LibreCAD/LibreCAD #2067: Polyline from Existing Segments transforms Arc into Ellipse

Hetman506, 2025-03-10:
Hello.

I noticed that in latest version of LibreCAD **Polyline from Existing Segments** tool transforms Arcs into Ellipses.

Steps to reproduce:
- create rectangle with rectangle tool
- fillet corner
- explode - Arcs are Arcs
- create polyline from existing segments,
- explode - Arcs are now Ellipses

Version: 2.2.2_alpha1-270-g8e25e848<br/>Compiler: GNU GCC 11.2.0<br/>Compiled on: Mar  1 2025<br/>Qt Version: 6.6.3<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

sand1024, 2025-03-10:
@Hetman506 yes, that's quite possible in that alpha, yet with later update I suppose this behavior will gone. 

@dxli -  yes, this behavior is due to experimental introduction of elliptic arcs as part of polyline. 

In my last PR I've disabled this behavior as it contradicts with overall polyline behavior and related actions. 

I suppose that for better support of fonts rendering and scaling, it's necessary to create a special version of polyline that will include elliptic arcs segments (non-persistent) and be used used within letters only.

Hetman506, 2025-03-20:
Everything seems to work on latest version :)

Version: 2.2.2_alpha1-294-ge6f74e4e<br/>Compiler: GNU GCC 11.2.0<br/>Compiled on: Mar 19 2025<br/>Qt Version: 6.6.3<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

sand1024, 2025-04-10:
@dxli 
Actually - no, it should not be closed. The issue with elliptic arcs in polylines still exists. 

If UCS is not rotated - everything is fine: 

[image]

However, if UCS has rotation - fonts looks ugly (screens are from the latest master). 

[image]

So there is an issue for rendering elliptic segments there. 

However, in wider sense, fixing the rendering only does not solve the issue with fonts scaling. 

As I've mentioned earlier, adding ellipse to polyline is actually a really bad idea. 

The reason is the following: 

Ellipse is non-standard extension of polyline, and ellipse segment can't be saved/stored in DXF.  
While from the first glance that's ok (as such polyline is just a part of letter rendering) - actually - that is not ok at all. 

One could make such polyline to be independent entity easily. 

First, the user may invoke "Explode Text to Letters" command first, and so each letters will be separate block. 

After that - it is possible to apply "Explode" command - and ta-dam! all polylines are not part of letter block and they are on their own. 

Result of saving such drawing is, of course, incorrect. 

So the correct way to handle such a situation - is introduce a special type of entity, create it on the fly during the creation of letter block, and take care of it additionally if such letter block is exploded. 

It's not still clear how to handle such Explode operation for such a polyline - yet most probably, the polyline that contains elliptic arcs should be also exploded to individual segments.

sand1024, 2025-04-17:
@dxli 
I'm not sure so far how the font file should be stored, but it seems that if .lff is saved after opening it, the elliptic segments are dropped (on saving) and so the letter polyline contains only vertexes and lines during next opening of that file. 

It seems that saving fonts files should be also adjusted to avoid such effects. Something like converting elliptic arc to cycle arc on save... or replacing elliptic arc by line interpolation... 

I think it even the opposite approach potentially may work (yet definitely this should be measured) - like doing line interpolation of arc segments on letter block polyline creation. Yet that might affect letter shape, especially for large scales... 

Actually, the implementation of fonts historically is quite a straightforward..