LibreCAD/LibreCAD #2069: MText not showing up until application (or file) reloaded

Daeraxa, 2025-03-13:
- Select MText tool, leave settings as default and add text
- Nothing shows under cursor on the screen
- Click and nothing seems to happen (cmd window changes to and shows `specify insertion point`)
- Reload LibreCAD (or close and re-open file) and the MText shows up

[image]

Version: v2.2.1.1
Compiler: GNU GCC 12.2.0
Compiled on: Feb 18 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009

dxli, 2025-03-13:
I couldn't reproduce this in Linux.

Daeraxa, 2025-03-13:
I was able to replicate this on a different Windows 10 machine, exactly the same behaviour.

Finnishsailor, 2025-05-05:
I have the same issue with Win 10.

DeverellManning, 2025-07-08:
I also have this issue on Windows 10.

rfsjim, 2025-07-16:
I can replicate this issue with Windows 11:
v2.2.1.1
Compiler: GNU GCC 12.2.0
Compiled on: Feb 18 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009

Saving, closing and reopening the file will make the mtext load.

dxli, 2025-09-17:
@SailorJon 

Is it possible for you to test this issue on windows 11? I currently don't have access to windows 10, but have a win11 virtual machine.

I have tested 2.2.1 and the master branches with my virtual machine win11:

1. start LibreCAD;
2. Start MText;
3. Choose font "unicode", and text="ABCD" in the MText dialog;
4. Close the dialog, and click to insert text.

Expected results: text is inserted and visible;
Actual results: text is inserted and visible.

Just wondering whether this is an issue limited to windows-10.

SailorJon, 2025-09-17:
Thanks for responding.
Sorry, I have resisted Windows 11!
Above, I find two posts, one on Windows 11 home, the other on Windows 11 Pro. Both experience the same fault:
@mikenoname-github 
@rfsjim 
So I conclude that both Windows operating systems expose this glitch in a clean install of  2.2.1, whereas a clean install of 2.2.0 on Windows 10 works wonderfully.
As a clean install was essential to remove this flaw, this may give a clue to coders in their efforts to resolve it.

rfsjim, 2025-09-18:
Have tested `MText` in `Version: v2.2.1.2` the work flow suggested by @dxli works as expected

1. Start LibreCAD
2. Select MText
3. *Choose font* - any font and text in the MTet dialog
4. Close the dialog and click to insert text

The *choose font* drop down is blank, if left blank it will only display the font after the application or file is reloaded.

Is it possible to have a default value in the font dropdown box?

SailorJon, 2025-09-18:
I have found that a clean install of this version:
Version: **2.2.0.2**
Compiler: GNU GCC 13.1.0
Compiled on: Jul 29 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)
Works fine on Windows 10 fully updated. The text dialog boxes offer me basic Latin, which works fine.
I cannot get the latest version to resolve this problem, despite fresh clean installation and trying various options. As I cannot see where text is located, I will have to stick with the previous version.
Thank you all very much for your attention to this matter.

dxli, 2025-09-21:
@SailorJon 

I found it's a bug tied to GUI font selection for both Text and MText:

1. If user doesn't select a font in the dialog, a default font is used, but the default font names have been changed long ago, so invalid default font names are used;
2. The invalid default font causes the selection displayed as empty in the dialog;
3. If the end user doesn't manually fix the empty selection, invalid font is used, so no text/mtext is displayed;
4. When the saved Text/MText is opened, the font resolution is not done through the dialog, and default font is used and the text is rendered;
5. Finally, this bug has nothing to do with Win10, and the proper way to reproduce is to cleanup any LibreCAD settings before starting LibreCAD, and avoid selecting a valid font in Text/MText creation dialogs.

Some cleanup is needed for the master branch, so a fix is only pushed for 2.2.1.