LibreCAD/LibreCAD #2070: Exporting visible or selected layers crashes the application

Daeraxa, 2025-03-13:
- Go to the layer list
- Select a layer
- Right click > Export Selected Layer(s)
- Application hangs for a second then closes

Same happens if exporting visible layers.

Unrelated to the crash but it seems odd that the menu items are only available in Layer List and not in the Layer Tree.

I wasn't sure how to obtain any crash or error logs, happy to provide if told how to get them.

Version: v2.2.1.1
Compiler: GNU GCC 12.2.0
Compiled on: Feb 18 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009

dxli, 2025-03-13:
I cannot reproduce this issue in Linux.

Could anyone help test in windows?

Daeraxa, 2025-03-13:
I was able to replicate this on a different Windows 10 machine, exactly the same behaviour.

sand1024, 2025-03-13:
@Daeraxa 

> Unrelated to the crash but it seems odd that the menu items are only available in Layer List and not in the Layer Tree.

Yes, this is a good point. I don't remember why the export was not added to Layer Tree, but probably it should be there indeed. 

Could you please create a separate issue for this?

sand1024, 2025-03-13:
@Daeraxa 

As a temporary workaround, please go to the Application Preferences dialog -> Defaults. 

And make sure that "Don't use native OS file open dialog" checkbox is CHECKED, and save settings. 

With this setup - the export should work.

Daeraxa, 2025-03-13:
Yup that works now, thank you!

dxli, 2025-03-14:
@Daeraxa 

A fix should be ready soon.

> Yup that works now, thank you!

dxli, 2025-03-14:
@sand1024 

Looks like we need to fix in both branches: master and 2.2.1.

I have assigned this one to you. Feel free to assign it back to me.

sand1024, 2025-03-15:
@dxli 
ok

closed 2025-03-23 by dxli