LibreCAD/LibreCAD #2071: Add "Export x Layer(s)" options to Layer Tree

Daeraxa, 2025-03-13:
The two items at the bottom of the Layer List context menu, `Export Selected Layer(s)` and `Export Visible Layer(s)`, are not present in the Layer Tree.

[image]

[image]

sand1024, 2025-04-23:
@Daeraxa
Should be fixed in last update after PR https://github.com/LibreCAD/LibreCAD/pull/2114 (in 2.2.2 alpha branch - continuous build)

Please check and close if completed.

Daeraxa, 2025-04-23:
I managed to export a single layer to a drawing but its hard to try to evaluate much else as the application keeps crashing and throwing errors before I can do anything.

Daeraxa, 2025-04-24:
Sorry I didn't have enough time after commenting that as I was going to create new issues as they are unrelated to this function. I'll create them as soon as I can.

closed 2026-09-26 by sand1024