LibreCAD/LibreCAD #2076: not displaying fully at 1080P resolution

BLSKWOLF, 2025-03-17:
My system is Ubuntu 22.04, and my screen resolution is set to 1920x1080. When I launch LibreCAD, the entire application window isn't visible; it's cut off. The user interface elements at the bottom of the window are compressed and overlapping. Switching to fullscreen mode doesn't resolve the problem; the display remains the same. When in windowed mode, I can only adjust the window width, but not the height.

[image]

sand1024, 2025-03-17:
@BLSKWOLF 
yes, I also suppose that the issue is related to right docking widget.  Please try hide/redock them. You can move widget by dragging on their title, btw.  

Of course, this window height is small enough to fit everything at the right, so it's necessary to hide some widgets. 

@dxli 
Actually, it seems the issue with default layout (one after first start) is quite common... 

And it's a kind of a paradox - for small screen resolutions, there is not enough space on the screen, for larger screen resolution - there is lots of empty unusable space on the screen. 

Probably it is necessary to expand the first load screen and add a kind of intro/setup wizard for the layout, so that will let the user to setup screen layout for it's own screen size?   I'm trying to remember in which software I've seen something similar... 

I actually doubt that we could find the default layout that will work reliable for all screen resolution and be informative to the user.

BLSKWOLF, 2025-03-18:
@sand1024 @dxli Thanks, this works! The bottom bar that was cut off is now visible. I remember other programs with this problem usually using either UI scaling or scrollbars.

closed 2025-03-20 by BLSKWOLF