LibreCAD/LibreCAD #2085: Continuous Build - broken for WIn?

sand1024, 2025-03-23:
@dxli 
Hi, 

It seems that something is not fully correct with the latest continuous build... 

[LibreCAD-2.2.2_alpha1-307-g281e9d02-win64.exe](https://github.com/LibreCAD/LibreCAD/releases/download/2.2.2_alpha-latest/LibreCAD-2.2.2_alpha1-307-g281e9d02-win64.exe)

I've installed it, and on a first and subsequent run got a message that entry point is not found, as follows: 

[image]

Could you please take a look?

dxli, 2025-03-25:
is the latest build working now?

I don't have a ready wndows machine to test this one.

sand1024, 2025-04-04:
@dxli 
Well, yes and no. 
I've downloaded installed the latest build (354). 
It starts, no error regarding entry points. 

However, it crashes on start - at the moment of the crash, the splash screen is still visible. 
The log file is clear, the last message is related to alias file, yet that's ok as I got the same message even during running LC from IDE. 

[image]

I've also checked that my local LC settings do not interfere with build build version - so I've cleared all local settings and restored them to default value. 

Started LC from build again - the fist run dialog was shown, after closing the dialog - main window with widgets were created (default layout). However, I still got a crash after that, splash screen is visible at the moment of crash :( 

Also, I've noticed that this build does not include custom icon engine - so I've copied my local copy to the installation folder, run again - the result is the same, a crash while splash screen is visible. 

I've checked log with debugging log level  - it is also not too informative: 

[image]

P.S. 
1) probably it's good to add some separation between the set of settings - ones for development and production ones - at least, they will not clash (using some macro to switch or so) - just to note this.  
2) QT - hm.. ok, I'll also upgrade to 6.9.0 and check. 
3) as for MSVC... I don't have it installed, probably it's also necessary to check. Yet I've seen MSVC last time years ago, that may take some time... 
4) CMake - I use CMake-based build in CLion, so cmake shouldn't be an issue, I suppose.

sand1024, 2025-04-04:
@dxli Oh, ok. 
The only thing - what do you mean by "standard data file path from QT"? Some specified location of QT? But under win it may be anywhere.  Or just some path with the installation layout? 

I remember that I've also seen some strange behavior with checking whether file is readable/writable. 

at least for something line iconsDir.isReadable();
 writingStyleEnabled = iconsDir.isWritable();

I got false for existing dir and at that time decided to return to this later on..

sand1024, 2025-04-05:
@dxli
LibreCAD-2.2.2_alpha1-360-g4dcf46266-win64.exe - works fine now, no crash on start.  

Of course, the functionality for changing icon colors does not work. That is understandable, however- as there is no dll with icon engine plugin. 

Interesting, yet even copying one that I've built locally does not work - most probably due to difference in QT version (it hardly due to the difference between mingv/mvcs builds, I suppose). 

Yet except this thing, it seems as for other features - the build is quite operational.

dxli, 2025-04-05:
@sand1024 

The alias crash was introduced by me, and the fix was trivial.

Yes, the plugins are not supported by CMake yet. Since I currently have no access to windows, could you add them in the CMakeFiles.txt? It might be faster to actually do it in Windows, instead of the github CI.

I was trying to add the ttf2lff tool as well, but didn't finish that yet.

sand1024, 2025-04-06:
@dxli 

Hm... I've checked out the latest version of CMakeLists.txt from master - and have a question regarding it. 

There are a couple of variables there: 
${FREETYPE_LIBRARIES}
${FREETYPE_INCLUDE_DIRS}

I suspect they are for FT_Library, but which version is expected? I didn't find the mention of this in the documents describing build process.

dxli, 2025-04-06:
The library is only needed for the ttf2lff tool, which is actually independent from the LibreCAD codebase.

u2fly, 2025-04-14:
Latest EXE also missing users plugins (for "Plugins" menu) and built-in resources (fonts, patterns, part library)

u2fly, 2025-04-17:
Tested latest EXE under Wine on Linux — it freezes on adding Arc Length dimension

sand1024, 2025-04-17:
@u2fly yes, freezes on adding Arc Length dimension confirmed. 
Could you please create separate ticket for this?

sand1024, 2026-09-26:
the issue is for outdated codebase, not exists now

closed 2026-09-26 by sand1024