LibreCAD/LibreCAD #2086: Crash when inserting from the library

dellus, 2025-03-24:
When I try to insert something from the library LC crashes/exits at the moment I specify the insertion point.
LibreCAD 2.2.2_alpha1-294 and -307 Linux appimage. -284 OK.
Puppylinux Fossapup.

dxli, 2025-03-25:
this one can be reproduced with AppImage, but not from local builds.

dxli, 2025-03-26:
I will keep this one open for some time. The reason: the fix is not logical to me, but it did fix the symptom: inserting a block from the library browser doesn't crash anymore with apooimage.

I am wondering whether there's an undefined behavior involved, as some pretty safe changes caused a crash.

dellus, 2025-03-27:
Sorry, for me the issue is not fixed in  2.2.2_alpha1-230 and -232 appimage. It still crashes when inserting a block from the library browser.

dxli, 2025-03-28:
@dellus 

I couldn't reproduce this issue with build 2.2.2_alpha1-341

I still feel this issue is caused by undefined behaviors of C++, so it only happens with the AppImage, but not with local builds. 

If it still crashes for you, I will review the compiler warnings. It would be different to find the actual cause though.

dellus, 2025-03-28:
Sorry, no joy in 2.2.2_alpha1-341for me either.

dxli, 2025-03-28:
@dellus 

is it possible for you to mount the appimage, and run librecad in gdb to find the backtrace? (command bt in gdb. of course, any tool to find the call stack is good).

As I couldn't reproduce this issue, hard to guess.

dellus, 2025-03-30:
I have experienced now that the crash happens randomly. When I tried 341 again to my surprise it didn't crash. But after closing and restarting it happened again. I went through several cycles with versions 294 to 341 and it was the same, some times it crashed and sometimes not. If it didn't crash at the first insert it wouldn't at following inserts.

I am not a developer. I have succeeded to install gdb, but I don't know how to run it. What does it mean to "mount an appimage"? I found no way to do this in my fossapup linux. I can mount a drive. 
What is the command to run a librecad.appimage from console?

dxli, 2025-03-31:
@dellus 

Thanks a lot. It helps a lot to know it crash somewhat randomly. The likelihood of some C++ undefined behavior is more likely with that information.

I have pushed a fix in master (build 346). It works for me so far. The commit fixes a memory leaking issue.

Please let us know if it still crashes.

dellus, 2025-03-31:
In -347 indeed I couldn't trigger a crash in several attempts. Thanks.
I leave it up to you to close the issue if you consider it as solved.

closed 2025-04-02 by dxli