LibreCAD/LibreCAD #2090: Dark theme: layer, block, pen, and other toolbars unreadable (white font on white bg)

jirislaby, 2025-04-01:
As can be seen:
[image]

I cannot read layers, pens and others. It is white font on white background (while using system's dark theme).

I use a git snapshot: f4c1d0038b61b89b6dc046a71f3db1e224fb4dae with qt6 (on wayland).

sand1024, 2025-04-01:
yes, there was an explicit setting for CSS that controls backgrounds. 
This will be fixes in the next PR

sand1024, 2025-04-23:
@jirislaby 
This is fixed in #2114, please check the latest 2.2.2 alpha

jirislaby, 2025-04-24:
`Layers` is fine.

`Pen Palette` and `Layers Tree` still affected.

sand1024, 2025-04-24:
@jirislaby 
Oh, thank you for you feedback. Yep, it was last minute pre-merge issue with these widgets, sorry. Will be fixed anyway.

closed 2026-09-26 by sand1024