LibreCAD/LibreCAD #2094: [git snapshot] Printing does not respect print preview selection

jirislaby, 2025-04-02:
If I select what to print in _Print Preview_:
[image]

It is not completely respected and the printed content (be it to pdf or to printer) is always from `[0, selected_y]`:
[image]

I.e. `x` is always `0`, apparently.

I run f4c1d0038b61b89b6dc046a71f3db1e224fb4dae. This is a regression from 2.2.1.1.

dxli, 2025-04-02:
@sand1024 

we have lots of trouble with bounding boxes. Probably due to CalculateBorders() caching.

sand1024, 2025-04-02:
@dxli  

Hm...  I don't think caching is an issue there - at least, I suppose it should be updated as part of generic update().  

However, at the moment the code in master that related to bounding box is broken in general - please check my comment regarding the clipping I've made for one of commits...  

Please review my comments within this thread:

https://github.com/LibreCAD/LibreCAD/commit/cc6437f589adcdb9c8e8b4c0d5986d0646e067d9#r153917209

Here I've also refer to the clip that illustrates the issue with box calculation and so with clipping. 

I think it's related to changed logic of calculation  there :( 

I'm absolutely sure (100%) that such clipping issue was not present in original implementation and PR (you can just test on PR code or using the branch from which PR was committed - https://github.com/sand1024/LibreCAD/tree/UCS), so I suspect that somewhere there is a difference in results between original/modified algorithms. Please review.

sand1024, 2025-04-02:
@jirislaby Could you please share the DXF with such an issue?

jirislaby, 2025-04-03:
> [@jirislaby](https://github.com/jirislaby) Could you please share the DXF with such an issue?

That would be the one from https://github.com/LibreCAD/LibreCAD/issues/2095#issue-2965232148.

u2fly, 2025-04-21:
Another case: if center drawing on page, it outs in PDF centered vertically, but aligned to the left paper edge.

LibreCAD-2.2.2_alpha1-398-g65f3c5fa7.AppImage