LibreCAD/LibreCAD #2095: [git snapshot] Printing does not respect line widths

jirislaby, 2025-04-02:
When I print [zahonky.tar.gz](https://github.com/user-attachments/files/19561790/zahonky.tar.gz) (sorry, I had to pack it, github does not allow `.dxf`), it is barely readable -- as if the lines were 0.01 mm.

I run https://github.com/LibreCAD/LibreCAD/commit/f4c1d0038b61b89b6dc046a71f3db1e224fb4dae. This is a regression from 2.2.1.1. If I go back, it is printed as set up.

dxli, 2025-04-05:
I cannot reproduce this one.

[Issue2095.pdf](https://github.com/user-attachments/files/19618652/Issue2095.pdf)

jirislaby, 2025-04-23:
Hmm, now even 2.2.1.1 renders very thin lines. Both to pdf and printer. Despite I set 2 mm for some layer. Do I have some general switch switched?

jirislaby, 2025-04-23:
Ah, yes, `Apply print scale to line width`, but I cannot turn it off for the printer, apparently (it re-enables itself while printing).

Ah, if I tick `fixed`, it does not re-set. That's a weird behaviour.

jirislaby, 2025-04-23:
I think I didn't test `fixed` in the git snapshot. Anyway, what does this `Apply print scale to line width` mean at all and why it produces unreadable output?