LibreCAD/LibreCAD #2096: [git snapshot] MText dialog contains garbage

jirislaby, 2025-04-02:
If I open MText dialog:
[image]

It contains garbage: `ÃÂedkev` (hex: `c3 83 c2 85 c3 82 c2 98  65 64 6b 65 76`). This should have been `ředkev` (`c5 99 65 64 6b 65 76`). Perhaps some utf8 input considered as some 8859-1 and `0xc5 0x99` is reencoded into ut8 as `u0xc383 u0xc285`?

In anyway where does the value come from? It is not the last one. Neither is font height remembered.

I run https://github.com/LibreCAD/LibreCAD/commit/f4c1d0038b61b89b6dc046a71f3db1e224fb4dae. This is a regression from 2.2.1.1.

jirislaby, 2025-04-02:
I have something like this in my mind:
```bash
> echo -n ř | iconv -f ISO-8859-1 -t utf8 | hexdump -C 
00000000  c3 85 c2 99                                       |....|
```

jirislaby, 2025-04-02:
This might be it:
https://github.com/LibreCAD/LibreCAD/commit/05c0f3bd865aae21f453965a504d215a63a02782
```diff
@@ -186,34 +184,35 @@ void QG_DlgMText::setText(RS_MText& t, bool isNew) {
         wPen->hide();
         lLayer->hide();
         cbLayer->hide();
-        RS_SETTINGS->beginGroup("/Draw");
+        LC_GROUP("Draw");
+        {
             //default font depending on locale
             //default font depending on locale (RLZ-> check this: QLocale::system().name() returns "fr_FR")
             QByteArray iso = RS_System::localeToISO(QLocale::system().name().toLocal8Bit());
 //        QByteArray iso = RS_System::localeToISO( QTextCodec::locale() );
             if (iso == "ISO8859-1") {
-             fon = RS_SETTINGS->readEntry("/TextFont", "normallatin1");
+                fon = LC_GET_STR("TextFont", "normallatin1");
             } else if (iso == "ISO8859-2") {
-             fon = RS_SETTINGS->readEntry("/TextFont", "normallatin2");
+                fon = LC_GET_STR("TextFont", "normallatin2");
             } else if (iso == "ISO8859-7") {
-             fon = RS_SETTINGS->readEntry("/TextFont", "greekc");
+                fon = LC_GET_STR("TextFont", "greekc");
             } else if (iso == "KOI8-U" || iso == "KOI8-R") {
-             fon = RS_SETTINGS->readEntry("/TextFont", "cyrillic_ii");
+                fon = LC_GET_STR("TextFont", "cyrillic_ii");
             } else {
-             fon = RS_SETTINGS->readEntry("/TextFont", "standard");
+                fon = LC_GET_STR("TextFont", "standard");
             }
-        height = RS_SETTINGS->readEntry("/TextHeight", "1.0");
-        def = RS_SETTINGS->readEntry("/TextDefault", "1");
-        alignment = RS_SETTINGS->readEntry("/TextAlignment", "1");
+            height = LC_GET_STR("TextHeight", "1.0");
+            def = LC_GET_STR("TextDefault", "1");
+            alignment = LC_GET_STR("TextAlignment", "1");
             //letterSpacing = RS_SETTINGS->readEntry("/TextLetterSpacing", "0");
             //wordSpacing = RS_SETTINGS->readEntry("/TextWordSpacing", "0");
-        lineSpacingFactor = RS_SETTINGS->readEntry("/TextLineSpacingFactor", "1");
-        str = RS_SETTINGS->readEntry("/TextString", "");
+            lineSpacingFactor = LC_GET_STR("TextLineSpacingFactor", "1");
+            str = LC_GET_STR("TextString", "");
             //shape = RS_SETTINGS->readEntry("/TextShape", "0");
-        angle = RS_SETTINGS->readEntry("/TextAngle", "0");
+            angle = LC_GET_STR("TextAngle", "0");
             //radius = RS_SETTINGS->readEntry("/TextRadius", "10");
             // leftToRight = RS_SETTINGS->readNumEntry("/TextLeftToRight", 1);
-        RS_SETTINGS->endGroup();
+        }
     } else {
         fon = text->getStyle();
         setFont(fon);

```

dxli, 2025-04-04:
@jirislaby 

No, probably not qg_dlgtext. we have Text and MText, and MText uses qg_dlgmtext.

dxli, 2025-04-05:
@jirislaby 

The issue is probably due to an incorrectly saved text.

Please try the latest build after my fix. Insert a new MText with correct text, and start the MText command again.

jirislaby, 2025-04-07:
> Please try the latest build after my fix. Insert a new MText with correct text, and start the MText command again.

Yes, now it properly stores the last text (and even its  height).

closed 2025-04-07 by jirislaby