LibreCAD/LibreCAD #2099: Win64 installer exits with error: A referral was returned from the server.

CloudInterfacesLLC, 2025-04-03:
OS: Win 11 (Build 26100)
LibreCAD Windows release: LibreCAD-v2.2.1.1-win64.exe
SHA256 Hash: C475163F5B3398482D10C567A620094DA692376311173B658F11DBBE4CAF1DEF

[image]

The error dialog reads: A referral was returned from the server.
I can also reproduce this fatal error on v2.2.1 stable build for win64.
I don't see any events logged. I don't see logs generated by the installer.

CloudInterfacesLLC, 2025-04-03:
To be clear, the installer was being run with administrator privileges.
I'm not willing to disable crypto and PKI signature checks, just for this one application, if that's the failure mode?

dxli, 2025-04-03:
@CloudInterfacesLLC 

This is a permission problem, not an installer issue.

Your desktop requires permissions from network for installation.

CloudInterfacesLLC, 2025-04-04:
Can you be more specific on the network permissions? I've scripted firewall configurations.
I use a VPN for everything I do online, and other installers don't fatal error and stop.

CloudInterfacesLLC, 2025-04-04:
> This is a permission problem, not an installer issue.
> Your desktop requires permissions from network for installation.
  
[@dxli](https://github.com/dxli)
Can you be more specific on the network permissions? I've scripted firewall configurations.
I use a VPN for everything I do online, and other installers don't fatal error and stop.

dxli, 2025-04-04:
@CloudInterfacesLLC 

https://learn.microsoft.com/en-us/answers/questions/1519115/i-have-a-problem-referral-was-returned-from-the-se

CloudInterfacesLLC, 2025-04-04:
[@dxli](https://github.com/dxli)  
Thanks that was unhelpful and rude for no reason.

dxli, 2025-04-04:
@CloudInterfacesLLC 

Sorry for any misunderstanding.

I am a developer of LibreCAD, and my understanding is that the error is not due to LibreCAD. However, I'm not an expert on Windows Management. The best thing I could point to you is the direction of Windows Management.

At a free software, we just want to improve the software and help people.

closed 2025-04-04 by dxli

CloudInterfacesLLC, 2025-04-05:
[@dxli](https://github.com/dxli)  
Well thanks dude, I'm glad I spent my time telling you that your software, and only your software, does not install.