LibreCAD/LibreCAD #2100: LibreCAD Appimage doesn't open under Wayland

38github, 2025-04-06:
I believe a lot of users would think something is wrong and maybe give up on LibreCAD since the it doesn't open. I have to execute `QT_QPA_PLATFORM=xcb ./LibreCAD-v2.2.1.1-10-gd57b53e75-x86_64.AppImage` in a terminal for it to open. A library that is not bundled into the Appimage?

dxli, 2025-04-06:
@38github 

Thanks a lot for troubleshooting.

It may also help to use snap or flatpak packages.

dxli, 2025-04-14:
@38github 

AppImage still fails for me on Wayland. Feel free to help here.

u2fly, 2025-04-18:
Version: 2.2.2.3-alpha<br/>Compiler: GNU GCC 13.3.0<br/>Compiled on: Apr 17 2025<br/>Qt Version: 6.4.2<br/>Boost Version: 1.83.0
System: Debian GNU/Linux 12 (bookworm)

Now runs, but shows networking warning message on startup

```
$ ./LibreCAD-2.2.2_alpha1-392-g68ffc982d.AppImage
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
Broken filename passed to function
Gtk-Message: 06:29:28.163: Failed to load module "canberra-gtk-module"
...
qt.tlsbackend.ossl: Failed to load libssl/libcrypto.
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslSocket
qt.network.ssl: The backend named "cert-only" does not support TLS
qt.network.ssl: QSslSocket::connectToHostEncrypted: TLS initialization failed
```

u2fly, 2025-04-19:
> Now runs, but shows networking warning message on startup

Also in `LibreCAD-2.2.2_alpha1-394-g522ec60a8.AppImage`
- has no bundled translations and plugins
- fonts/library/patterns are bundled, but not recognized by LibreCAD

u2fly, 2025-04-20:
@dxli 

CI failed before `update-appimage.sh` run
- https://github.com/LibreCAD/LibreCAD/actions/runs/14563621552/job/40849942414

u2fly, 2025-04-21:
@sand1024 

Check

- https://github.com/LibreCAD/LibreCAD/actions/runs/14563612448/job/40849924621#step:5:747