LibreCAD/LibreCAD #2103: Settings files do not save in Other Settings Locations path if changed

Questions2000, 2025-04-08:
Little confused, I made this new post just in case these items are not connected, but this post is similar to the following post
#2018
If you guys could discuss and let me know if I should close any of the two posts.

### Expected Behavior
For settings files to save in user decided path location set in Application Preferences - Paths tab

### Observed Behavior
I recently noticed that if you go into `Application Preferences - Paths` and change `Other Settings Locations` from default path which is
`/home/<username>/.local/share/LibreCAD/LibreCAD/`
to a different location, such as
`/home/<username>/.config/LibreCAD`

Files such as 
`workspaces.lcws`
`librecad.alias`
still seem to get saved in the default location.

Unless I am overlooking the purpose of changing path, it appears that users can now change settings path location, but it does not appear to change where LC is saving the settings files.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-348-gce9db2175.AppImage

sand1024, 2025-04-08:
@Questions2000 
Oh... that's good point. Well, that's strange, I'll take a look on this.

sand1024, 2025-04-08:
@Questions2000 
yes, this is a bug, if will be fixed in next PR

sand1024, 2025-04-23:
@Questions2000 
This is fixed as part of #2114

closed 2025-05-20 by Questions2000