LibreCAD/LibreCAD #2104: Snap on entity conflicts with angle guide line

Questions2000, 2025-04-08:
### Expected Behavior
When drawing a line and holding "Shift" to be able to see the line indicator of line that is going to be drawn.
Should be able to draw lines regardless of if snap settings are met or not.
Should include "Snap On Entity"

### Observed Behavior
When drawing a line, if "Snap On Entity" snap is "on" and you hold "Shift" key to activate angle guide, the gray indicator line that shows the line to be drawn does not show up all the time.
Only shows line if there are other drawn entities in that direction where the line is being drawn.

### Steps To Reproduce
-Turn on "snap on entity" snap setting
-Do this is a blank drawing (important)
-Activate "2 Point Line" command
-Click 1st point
-Hold "Shift" key and add 2nd point to the right (will be horizontally placed in line with the 1st point)
-Keep holding "Shift" and now continue that line down by clicking adding a 3rd point vertically below that 2nd point.
When try adding this 3rd point you should notice that the indicator line does not show up and you can kind of see the angle guide glitching a little bit.

If you repeat these steps above, but before doing so draw a horizontal line below where that 3rd point is added, then when repeating these steps, vertical guide line shows up fine now that there is an object below that gets detected.

Should be able to keep snap settings "on" and still be able to draw lines.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-348-gce9db2175.AppImage

sand1024, 2025-04-08:
@Questions2000 

Well, I don't think this is a bug - it's rather a side effect of the snap and the question of priorities...   And if I correct, this situation occurs only if "snap to entity" is the only snap mode enabled. 

According to your scenario 
1) you've created first entity
2) during creation of the second entity, snap is performed to the only existing entity (the first on) that prevent Angle Snap. 

So here results of "snap to entity" has higher priority than Angle Snap. 

If, for example, "snap to entity" and "snap to grid" are both enabled - snapping is performed in the direction to angle. 

I'll check whether the case with snapping where only "Snap to Entity" is enabled may be improved, yet rather consider this as an improvement rather than an issue.

ezequielleonzybert, 2025-04-16:
This happens from a long time ago now. I tested this app like a year or two ago and this bug was one of the first issues I encountered on it.

ezequielleonzybert, 2025-04-16:
https://github.com/user-attachments/assets/42ecb221-0f8f-46f3-b873-8e10bc260a25

in that video I'm holding Shift while I draw a line from the mid top point to the mid bottom point.
The snapping wiggles if I move the mouse a little, it takes as priority the angle snapping instead of the mid point.

ezequielleonzybert, 2025-04-16:
@sand1024 
https://github.com/user-attachments/assets/b98b7b71-bebf-4fd8-a546-45b2ecd227bc
With free snap disabled the effect is even worst

ezequielleonzybert, 2025-04-16:
@sand1024

https://github.com/user-attachments/assets/0ace47df-738a-4b7d-adcb-4e967c778fec

With free snap disabled the effect is even worst