LibreCAD/LibreCAD #2105: Orthagonal line tool causes application to close instantly

Questions2000, 2025-04-08:
### Expected Behavior
To have LibreCAD stay open when tools are used :)

### Observed Behavior
When using Orthagonal line tool, LibreCAD closes instantly.

### Steps To Reproduce
-Draw a horizontal 2 Point line
-Activate the line tool called "Orthagonal"
-Click on the right end point of the horizontal line drawn
-Click mouse again after and you will notice that the instant you click that second time, LC closes instantly.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-348-gce9db2175.AppImage

dxli, 2025-04-08:
cannot reproduce with build 367

sand1024, 2025-04-08:
@dxli 
I can't reproduce it either, yet this action behaves a bit strange, so I'll check it deeper.

dxli, 2025-04-14:
The root cause, uninitialized pointer dereferenced.

Just anothe case, uninitialized primitives should be avoided.

closed 2025-05-10 by Questions2000