LibreCAD/LibreCAD #2108: Opening a newly created / stored drawing : Invalid objects removed: 1

hapees, 2025-04-14:
Hi,

with the most recent version (and at least the one before) the drawings are not opened correctly:

Version: 2.2.2_alpha1-389-g1cbc35dfd
<br/>Compiler: GNU GCC 14.2.1<br/>Compiled on: Apr 14 2025<br/>Qt Version: 6.8.2<br/>Boost Version: 1.83.0
System: Fedora Linux 41 (Server Edition)

A drawing with  4 lines does not get opened correctly. (attached)

zip content of attached file:
1. File **Debug-created-saved.dxf** is the original drawing.
- draw 4 lines
- save 

2. File **Debug-Loaded-Saved.dxf** :
- Open Debug-created-saved.dxf
-> Message _Invalid objects removed: 1_
- save (no changes were made)

[srg_20250414_Invalid_object.zip](https://github.com/user-attachments/files/19734760/srg_20250414_Invalid_object.zip)

Remark:
Older dxf files open with _Message _Invalid objects removed: NNN_ where NNN is a large number.

sand1024, 2025-04-14:
@hapees 
Yes, thank you - this is known regression, will be fixed in one of nearest updates.

dxli, 2025-04-14:
@sand1024 

Thanks! We are not afraid of regressions in 2.2.2. If tere's a chance to cleanup, we should proceed.

We still have lots of so called "God API", for example, in qc_applicationwindow, and qg_graphicview etc.

The problem was due to short term  convenience, helper methods were being added to the huge pools. Over time, it's getting harder and harder to decouple and redesign.

Decoupling from Qt is also important. Qt headers may interfere with c++ standard headers(quite subtle to fix, in past Qt related issues), or at least causing trouble in newer Qt versions. That's the reason that Qt headers shoulds be avoided in our headers, whenever possible.

sand1024, 2025-04-14:
@dxli 
Yes, currently qc_applicationwindow is a god-like object with lots of functions... However, I already was able to refactor it and make corresponding code to specific classes, so now it's not overly-coupled and there is no mix of concerns. 

As for headers - I'm also cleaned them for lots of classes, replacing them by forward declarations where possible. 
I'll commit that in next PR. 

As for QT... well, here the situation is harder, it's all over the codebase. But I really doubt that QT may be fully isolated and, say, other library may be used instead of it. With time, of course, some QT containers or so may be replaced to relaxing dependencies to QT, yet still, I suppose it will be partial.

sand1024, 2025-04-23:
@hapees 
Please re-test on the latest 2.2.2 alpha and either close the issue or provide a sample of dxf where the issue still persists.

closed 2026-09-26 by sand1024