LibreCAD/LibreCAD #2111: Arc end-points regression (Hang on DimArc)

sand1024, 2025-04-17:
Creation of DimARC hangs the application (infinite cycle) (2.2.2 master)

sand1024, 2025-04-17:
@dxli 
The change included to revision 32614b47fe46b77006e4da473cecaeef44f0fc32 in RS_Arc entity, changes the logic of end-point getter (from returning originally calculated value of the end point to cached value). This introduces quite a serious bug.   

The change is for 
```

RS_Vector RS_Arc::getEndpoint() const{
**// from**
    return data.center + RS_Vector::polar(data.radius, data.angle2);
**// to**
   return endPoint;
}
```

The function returns cached value, yet that value is updated only by call of calculateBorders() method.  Since the semantic of the getter was changed, the caller code was not updated cached endpoint values.  

This lead for various effects, but in lc_dimarc.cpp the infinite cycle occurs (around line 323): 

```
  while (((dimArc1->getEndpoint().x < cornerLeftX)   || (dimArc1->getEndpoint().x > cornerRightX) 
    ||      (dimArc1->getEndpoint().y < cornerBottomY) || (dimArc1->getEndpoint().y > cornerTopY))

    &&     (dimArc1->getAngle2() < RS_MAXDOUBLE) 
    &&     (dimArc1->getAngle2() > RS_MINDOUBLE))
    {
        dimArc1->setAngle2(dimArc1->getAngle2() + deltaOffset);
        dimArc1->calculateBorders(); // this is a fix 
    }
```

I've added necessary fixes for this issue where needed, and I'll commit them with next PR. Yet please test your changes more carefully before commit, especially for core areas. That issue is easily discoverable during the development just by checking places where setAngle()/getEndpoint()  are called and smoke testing corresponding logic.

u2fly, 2025-04-18:
@dxli 

With https://github.com/LibreCAD/LibreCAD/commit/4f0d7c15e26006d4d35b798ecb95c6d68dee7093 latest build not hangs on adding DimArc, but now hangs on first or next click or move of created DimArc by cursor.

There is nothing about hang in terminal output.

u2fly, 2025-04-18:
> With https://github.com/LibreCAD/LibreCAD/commit/4f0d7c15e26006d4d35b798ecb95c6d68dee7093 latest build not hangs on adding DimArc, but now hangs on first or next click or move of created DimArc by cursor.
>
> There is nothing about hang in terminal output.

UPDATE: This new hang happens if "Distance" snap mode active.

If turn off "Distance" snap before moving created DimArc there is no hang.

dxli, 2025-04-18:
@u2fly 

DimArc implementation is not mature.

Distance snapping is probably an independent regression through.

dxli, 2025-04-28:
@u2fly 

could you test this issue with the latest commit: 7baab9895 ?

We got some data model issue here, and 7baab9895 is a first step to clean up the data model.