LibreCAD/LibreCAD #2112: LibreCAD Font 2 (LFF format)

u2fly, 2025-04-17:
@dxli @sand1024

- https://github.com/LibreCAD/LibreCAD/issues/2067
> I'm not sure so far how the font file should be stored
> ...
> Actually, the implementation of fonts historically is quite a straightforward..

### Expected behaiour

LFF format has version identifier on the first line:

```
# Format:            LibreCAD Font 1
# Creator:           LibreCAD
...
```

Preserve `LibreCAD Font 1` format behaiour as is, and all the new changes introduce in `LibreCAD Font 2` format (LibreCAD should handle fonts according format version).

In the next format:
- add ellipse arcs/splines/etc. for defining shape;
- add to `C<refchar_ID>` optional parameters for referenced character offset (to move refchar origin from actual char origin) with scale option (refchar origin as scale center) — this especially useful for Unicode Spacing Modifier Letters, superscript and subscript like `²` is scaled and offset variant of reference character `2`;
- add optional `L<ligchar_ID>` operator to combine actual `<char_ID>` with preceding characters in text line (if they are `<ligchar_ID>`) into ligature, and in replace ligature character with single character ligature shape
- add support for Unicode Combining characters, zero-width joiner, etc.
   - https://en.wikipedia.org/wiki/Combining_character
   - https://en.wikipedia.org/wiki/Zero-width_joiner
- Handle recursions in LFF files (report to Cmd problematic characters IDs in used fonts)
  - https://github.com/LibreCAD/LibreCAD/issues/1837
- Update LFF formats spec
  - https://github.com/LibreCAD/LibreCAD/wiki/LFF-Font-Format

###  LibreCAD Font 2

```Shell
# Format:            LibreCAD Font 2
# Creator:           LibreCAD
...

[<char_ID>] <symbol>
# Copy reference characters shapes
C<refchar1_ID>;<offset_X>,<offset_Y>;<scale>
C<refchar2_ID>;<offset_X>,<offset_Y>;<scale>
# Own shape from polylines/lines/etc
...
# Ligature1 for <char_ID> with preceding <ligchar_ID>
L<ligchar_ID>
# Ligature1 shape
...
# Ligature2 for <char_ID> with preceding <ligchar1_ID> or <ligchar2_ID>
L<ligchar1_ID>,<ligchar2_ID>
# Ligature2 shape, with copy reference character shape
C<refchar3_ID>;<offset_X>,<offset_Y>;<scale>
...
# Ligature3 for <char_ID> with preceding <ligchar1_ID><ligchar2_ID> or <ligchar3_ID>
L<ligchar1_ID>,<ligchar2_ID>;<ligchar3_ID>
# Ligature3 shape
...
```

dxli, 2025-04-17:
@u2fly 

LFF was design by our CAD master allaz. It probably needs some new design, or taking some new development from other open source projects. At least, we should have some ideas on how to improve efficiency on font file formats and font polylines.

For Elliptic arcs. I think it's a good idea, if we can improve file IO support:

1, always regenerate polylines for fonts (text/mtext), obviate relying polyline support for elliptic arcs;
2. improve elliptic arcs handling in our polylines. Finalize features, whether ellptic arcs should be limited fonts.

sand1024, 2025-04-17:
@dxli 
Yes, indeed - the general fonts subsystem needs some facelifting...  It's quite messy and does not perform well. 

As for elliptic arcs in polylines  - while they are fine to be supported for fonts, I'm really doubt we could support it for polyline. 

I had to check in more details, yet as far as I remember, the DXF format itself does not support elliptic arcs as segments of polylines..  If this is still true, it will be challenged to produce compatible DXF files with such polylines (or, it will be necessary to so some approximations there)

u2fly, 2025-04-17:
> It's quite messy and does not perform well.

In render or parsing LFF file?

sand1024, 2025-04-17:
@u2fly  hm... it's a good question. In architectural design and rendering speed of them, I would say. So some improvements there are needed.

u2fly, 2025-04-19:
> It probably needs some new design, or taking some new development from other open source projects. At least, we should have some ideas on how to improve efficiency on font file formats and font polylines.

Create a font format based on G-Code, that supports Line, Arc, Circle, Cubic Bézier Spline and much more
- Line - https://marlinfw.org/docs/gcode/G000-G001.html
- Arc and Circle - https://marlinfw.org/docs/gcode/G002-G003.html
- Bézier Cubic Spline - https://marlinfw.org/docs/gcode/G005.html

Take a look on this SVG to G-Code converter (this and hints above may be used as logic for supporting SVG vector import in LibreCAD in DXF-friendly format)
- https://github.com/domoszlai/juicy-gcode#features

Hints
- Ellipse to G-Code (4 Arcs?, Elliptical Arc - 3 Arcs?) - https://en.industryarena.com/forum/how-do-i-gcode-ellipses--312772.html
- G-Code Arc to SVG Elliptical Arc - https://stackoverflow.com/a/71327272
  - https://www.w3.org/TR/SVG2/paths.html#PathDataEllipticalArcCommands
- Quadratic Bezier Curve to Cubic Bézier Curve (not Spline), and vice versa (also https://github.com/LibreCAD/LibreCAD/issues/322#issuecomment-4865555 and https://github.com/LibreCAD/LibreCAD/issues/513#issuecomment-93000769)
- https://commons.wikimedia.org/wiki/File:Quadratic_to_cubic_Bezier_curve.svg
> A cubic Bézier curve (yellow) can be made identical to a quadratic one (black) by
> 1. copying the end points, and
> 2. placing its 2 middle control points (yellow circles) 2/3 along line segments from the end points to the quadratic curve's middle control point (black rectangle).

### Support for OpentType-SVG, SVG Font, "single-stroke TTF"

PoC of single stroke font
- https://github.com/isdat-type/Relief-SingleLine
  - OpenType-SVG and SVG variants depend on SVG support in LibreCAD
  - "Single-stroke TTF" variant requires TTF support with special feature for "open outline" rendering

closed 2025-04-27 by u2fly