LibreCAD/LibreCAD #2113: Recent Continuous Release AppImage does not work and also increased size

Questions2000, 2025-04-19:
I had originally typed this out for the following version
LibreCAD-2.2.2_alpha1-389-g1cbc35dfd.AppImage
When I tried to launch this version it did not work.
I have now seen that some newer versions were released that I have not tested yet, I will report if these newer AppImages do not work either.
(Just thought about it, maybe new stuff has been added that requires fresh start of LC settings, which occurred in the past, I will also try this out)

However there is still question about increased size.
I also noticed that this AppImage is about 2.5 times more MB size
Typically 40MB
This version = about 100MB

Just noticed that newer versions are now 116MB

Just mentioning because not sure if this is due to that much new code, or if this is some kind of compiling issue that is occurring?

Thank You

u2fly, 2025-04-19:
> LibreCAD-2.2.2_alpha1-389-g1cbc35dfd.AppImage

Duplicate of https://github.com/LibreCAD/LibreCAD/issues/2100

dxli, 2025-04-19:
@u2fly 

For translations and bundled resources, it's straightforward to fix their appimage locations.

Could you help here:

closed 2025-04-19 by dxli

u2fly, 2025-04-20:
@sand1024 

> Could you help here:
> https://github.com/LibreCAD/LibreCAD/blob/master/CI/build-appimg.sh

Next commands bundled nothing to AppImage, I don't know why

```
cp unix/resources/plugins/*.so appdir/usr/lib/librecad/
cp -r unix/resources/qm appdir/usr/share/librecad/
```

From CI log (https://github.com/LibreCAD/LibreCAD/actions/runs/14540989853/job/40798788196), 

- *.qm files located in `unix/*.qm`, not in `unix/resources/qm`?
```
2025-04-18T20:01:26.7974926Z Updating '/home/runner/work/LibreCAD/LibreCAD/unix/librecad_tr.qm'...
2025-04-18T20:01:26.8141273Z Updating '/home/runner/work/LibreCAD/LibreCAD/unix/librecad_uk.qm'...
2025-04-18T20:01:26.8200334Z Updating '/home/runner/work/LibreCAD/LibreCAD/unix/librecad_zh_cn.qm'...
```

- its not clear where are  plugins *.so located, and so nothing been copied
```
2025-04-18T20:07:56.8675329Z + strip 'unix/resources/plugins/*.so'
2025-04-18T20:07:56.8688868Z strip: 'unix/resources/plugins/*.so': No such file
2025-04-18T20:07:56.8691342Z + cp unix/librecad appdir/usr/bin/
2025-04-18T20:07:56.8834834Z + cp 'unix/resources/plugins/*.so' appdir/usr/lib/librecad/
2025-04-18T20:07:56.8848304Z cp: cannot stat 'unix/resources/plugins/*.so': No such file or directory
2025-04-18T20:07:56.8850434Z + cp -r unix/resources/qm appdir/usr/share/librecad/
2025-04-18T20:07:56.8862185Z cp: cannot stat 'unix/resources/qm': No such file or directory
```

- fonts/library/patterns copied, but LibreCAD does not see them
- ```
2025-04-18T20:07:56.8907995Z + cp -r librecad/support/fonts appdir/usr/share/librecad/
2025-04-18T20:07:56.9118611Z + cp -r librecad/support/library appdir/usr/share/librecad/
2025-04-18T20:07:56.9734447Z + cp -r librecad/support/patterns appdir/usr/share/librecad/
```

File `../AppRun` not configured for use bundled `fonts/library/patterns` paths?
Or CMake issue flags issue?

Also, I has installed LibreCAD 2.2.1 via DEB-package system wide, so there are files on PC

- /usr/share/librecad/library/../*.dxf
- /usr/share/librecad/patterns/../*.dxf
- /usr/share/librecad/fonts/*.lff
- /usr/lib/librecad/*.so
- /usr/share/librecad/plugins/*.so
- /usr/share/librecad/qm/*.qm

But AppImage not see these system wide files too.

Questions2000, 2025-04-22:
I was able to try this version which now does open correctly.
LibreCAD-2.2.2_alpha1-394-g522ec60a8.AppImage
I was going to make a separate post about this, I noticed a change in GUI fonts in this version.

Nobody answered my question about the size increase of the AppImage.
Does anyone know if there is a reason for this?