LibreCAD/LibreCAD #2119: Use bold text for section header names in all Options based dialogs

Questions2000, 2025-04-22:
If does not look too overwhelming, maybe can use bold text for section header names in all the Options based dialogs.
Just to make section headers stand out a little bit more.

The Option based dialogs I am speaking of are ones such as....
`Application Preferences`
`Widget Options`
`Device Options` (only has one section header)
`Current Drawing Preferences`

Also example of what I am calling "section header names" would be in
Application Preferences - Graphic View (tab)
`General`
`Background and Selection`
`Draft Mode Marker`

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-394-g522ec60a8.AppImage

sand1024, 2026-09-27:
@Questions2000 
Yes, the boldness of the group is defined by the UI style. While it's possible to override them and set bold text manually - that's usually not recommended in UX. For custom theme, however, if will be possible to customize this.

closed 2026-09-27 by sand1024

Questions2000, 2026-10-01:
@sand1024
Saw that this as closed recently.
I am not sure I follow.
I have some questions that maybe you or someone else could answer about the topic?

----------------------------------------

QUESTION 1
So you are saying that this is not recommended for developers to apply bold text in coding?

For example
`Application Preferences` window
`Graphic View` tab
Contains section text such as....
`General`
`Overlay Selection Box`
`Entity Handle`
etc.

These section header text items would not be good practice to bold text by default?
Strange I have seen some applications use Bold text to define section header text in their Preferences/Options windows.
Really helps sections standout from each other.

----------------------------------------

QUESTION 2
I was also going to mention in a separate post but will just mentioned here because it is about making section stand out.
Currently thin / light colored lines are used to outline sections.
Maybe making lines darker / bolder could also help to define sections?

----------------------------------------

QUESTION 3
Custom Theme
You mentioned that this is available using a "custom theme"?
For Linux users, do you mean "Style Sheet" using Qt QSS to achieve this?  Or do you mean this only for Windows users?

I never thought about being able to customize how specific LC windows appear.  I remember QSS changing main window, some widgets, context menus, etc. but I do not remember how LC windows such as `Application Preferences` appear.
Are you saying that the "Section Header Text" (that I mentioned in example above) can be made bold and it will not affect any of the other text in LC?
Do you know the line of text that would be added to a .qss style sheet file to achieve this?

Thank You

sand1024, 2026-10-01:
@Questions2000 
1) no, the boldness of such elements is defined by the overall application UI style. 
It will not be consistent to mark them manually in specific dialogs. 

2) that's defined by UI style. soon there it will be possible to tune UI theme better, including typography, colors, metrics and appearance of some controls. 

3) custom theme is one that is not native (i.e. drawn by controls of underlying operating system). It's not related to QSS. 

"Fusion" theme is Qt provided and it is not native, as it's Qt library who do drawing of controls.

Questions2000, 2026-10-10:
@sand1024
Thank you for taking the time to answer each one of my questions.
Looking forward to see what will be able to be modified.