LibreCAD/LibreCAD #2127: QT errors on startup

Daeraxa, 2025-04-25:
Version: 2.2.2.3-alpha<br/>Compiler: Microsoft Visual C++<br/>Compiled on: Apr 23 2025<br/>Qt Version: 6.9.0<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

These errors show on startup. I'm having other issues relating to my files not displaying in the application and various crashes but unsure if they are caused by this problem.

[image]

```
WARNING: CPU random generator seem to be failing, disabling hardware random number generation
WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
C:/Program Files/LibreCAD/resources/qm
C:/Program Files/LibreCAD/resources/fonts
C:/Program Files/LibreCAD/resources/fonts
font: C:/Program Files/LibreCAD/resources/fonts/amiri-regular.lff:
base: amiri-regular
font: C:/Program Files/LibreCAD/resources/fonts/azomix.lff:
base: azomix
font: C:/Program Files/LibreCAD/resources/fonts/azomix_i.lff:
base: azomix_i
font: C:/Program Files/LibreCAD/resources/fonts/cursive.lff:
base: cursive
font: C:/Program Files/LibreCAD/resources/fonts/cyrillic_ii.lff:
base: cyrillic_ii
font: C:/Program Files/LibreCAD/resources/fonts/gothgbt.lff:
base: gothgbt
font: C:/Program Files/LibreCAD/resources/fonts/gothgrt.lff:
base: gothgrt
font: C:/Program Files/LibreCAD/resources/fonts/gothitt.lff:
base: gothitt
font: C:/Program Files/LibreCAD/resources/fonts/greek_ol.lff:
base: greek_ol
font: C:/Program Files/LibreCAD/resources/fonts/greekc.lff:
base: greekc
font: C:/Program Files/LibreCAD/resources/fonts/greekcs.lff:
base: greekcs
font: C:/Program Files/LibreCAD/resources/fonts/greekp.lff:
base: greekp
font: C:/Program Files/LibreCAD/resources/fonts/greeks.lff:
base: greeks
font: C:/Program Files/LibreCAD/resources/fonts/iso.lff:
base: iso
font: C:/Program Files/LibreCAD/resources/fonts/iso3098.lff:
base: iso3098
font: C:/Program Files/LibreCAD/resources/fonts/iso3098_i.lff:
base: iso3098_i
font: C:/Program Files/LibreCAD/resources/fonts/italicc.lff:
base: italicc
font: C:/Program Files/LibreCAD/resources/fonts/italiccs.lff:
base: italiccs
font: C:/Program Files/LibreCAD/resources/fonts/italict.lff:
base: italict
font: C:/Program Files/LibreCAD/resources/fonts/kochigothic.lff:
base: kochigothic
font: C:/Program Files/LibreCAD/resources/fonts/kochimincho.lff:
base: kochimincho
font: C:/Program Files/LibreCAD/resources/fonts/kst32b.lff:
base: kst32b
font: C:/Program Files/LibreCAD/resources/fonts/lc_opengost-ar.lff:
base: lc_opengost-ar
font: C:/Program Files/LibreCAD/resources/fonts/lc_opengost-br.lff:
base: lc_opengost-br
font: C:/Program Files/LibreCAD/resources/fonts/opengosttypea-regular.lff:
base: opengosttypea-regular
font: C:/Program Files/LibreCAD/resources/fonts/opengosttypeb-regular.lff:
base: opengosttypeb-regular
font: C:/Program Files/LibreCAD/resources/fonts/romanc.lff:
base: romanc
font: C:/Program Files/LibreCAD/resources/fonts/romancs.lff:
base: romancs
font: C:/Program Files/LibreCAD/resources/fonts/romand.lff:
base: romand
font: C:/Program Files/LibreCAD/resources/fonts/romanp.lff:
base: romanp
font: C:/Program Files/LibreCAD/resources/fonts/romans.lff:
base: romans
font: C:/Program Files/LibreCAD/resources/fonts/romansi.lff:
base: romansi
font: C:/Program Files/LibreCAD/resources/fonts/romant.lff:
base: romant
font: C:/Program Files/LibreCAD/resources/fonts/scriptc.lff:
base: scriptc
font: C:/Program Files/LibreCAD/resources/fonts/scripts.lff:
base: scripts
font: C:/Program Files/LibreCAD/resources/fonts/simplex.lff:
base: simplex
font: C:/Program Files/LibreCAD/resources/fonts/standard.lff:
base: standard
font: C:/Program Files/LibreCAD/resources/fonts/syastro.lff:
base: syastro
font: C:/Program Files/LibreCAD/resources/fonts/symap.lff:
base: symap
font: C:/Program Files/LibreCAD/resources/fonts/symath.lff:
base: symath
font: C:/Program Files/LibreCAD/resources/fonts/symbol.lff:
base: symbol
font: C:/Program Files/LibreCAD/resources/fonts/symbol_misc1.lff:
base: symbol_misc1
font: C:/Program Files/LibreCAD/resources/fonts/symbol_misc2.lff:
base: symbol_misc2
font: C:/Program Files/LibreCAD/resources/fonts/symeteo.lff:
base: symeteo
font: C:/Program Files/LibreCAD/resources/fonts/symusic.lff:
base: symusic
font: C:/Program Files/LibreCAD/resources/fonts/unicode.lff:
base: unicode
C:/Program Files/LibreCAD/resources/patterns
C:/Program Files/LibreCAD/resources/qm
C:/Program Files/LibreCAD/resources/library
C:\Users\<User>\Documents\<Blocks Path>
readAliasFile(): Command alias file: C:/Users/<user>/AppData/Roaming/LibreCAD/LibreCAD/librecad.alias
C:/Program Files/LibreCAD/resources/plugins
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
qt.core.qobject.connect: QObject::connect: signal not found in QC_ApplicationWindow
```

dxli, 2025-04-25:
@Daeraxa 

Could you uninstall and manually delete any leftover files? The "*1.dll" names are weird, as the expected plugin names should not have the tailing digit 1.

Daeraxa, 2025-04-25:
I uninstalled fully and re-installed the alpha and it doesn't seem to be doing this any more.

closed 2025-04-26 by Daeraxa