LibreCAD/LibreCAD #2129: Application crashes if menu windows opened with no active drawings

Daeraxa, 2025-04-25:
Version: 2.2.2.3-alpha<br/>Compiler: Microsoft Visual C++<br/>Compiled on: Apr 23 2025<br/>Qt Version: 6.9.0<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

Steps:
- Open LibreCAD, close any open drawings
- File > Open
- Application crashes
OR:
- Open LibreCAD, close any open drawings
- Options > Application Preferences
- Application crashes
OR:
- Open LibreCAD, close any open drawings
- Options > Keyboard Shortcuts
- Application crashes

sand1024, 2025-04-25:
@Daeraxa 
Hm.. that's strange, I could not reproduce such behavior under Win 10... 
Could you please try to export your settings and share export file? Settings may be exported in Application Preferences -> Paths tab, the button is Export Settings to File. 
Thank you.

Daeraxa, 2025-04-25:
I re-installed the alpha completely (I checked to see if this happens with the current release but it doesn't) and it is still happening. 

Settings file attached (with .txt added).
[LCSettings.lcs.txt](https://github.com/user-attachments/files/19917578/LCSettings.lcs.txt)

closed 2025-06-03 by dxli