LibreCAD/LibreCAD #2142: Color picker dialog - Show color in dialog when color names are used

Questions2000, 2025-05-09:
In any settings dialogs that contain color drop downs / color picker buttons
I would think any issues with these would all be the same, but I have noticed some differences/issues with how these work in certain settings dialogs.
Might not be important, but I will still try to point out any inconsistencies I come across.

### Expected Behavior
From what I have seen, when actual color names are used/typed into the drop down field (for example Red, Cyan, etc.), if you hit the color icon button to their right, the color picker dialog shows up and it automatically shows that specific color/code, which is the expected behavior.

### Observed Behavior
In `Application Preferences - Info Cursor` there are 4 color settings
I think by default they are set to
`Orange`
`Gray`
`Yellow`
`Cyan`
If you open up any of their color picker dialogs by hitting button, they all show the color/code for `Yellow`, instead of what they say they are in the drop down.

### Similar posts
https://github.com/LibreCAD/LibreCAD/issues/2141

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-415-gd06f27896.AppImage

sand1024, 2025-05-10:
@Questions2000 
I suppose that's related to the color naming. 

There are the set (quite a limited) of named colors. 
If the code of the color corresponds to one of such named colors - the name of the color is shown instead of the code. Otherwise, the code is shown. 

Or probably I've missed something and you actually point to something else?

Questions2000, 2025-05-14:
@sand1024 
? Not sure I follow, did you try what I mentioned?
If you go into
`Application Preferences - Info Cursor` tab
There are 4 colors settings which are default
`Orange`
`Gray`
`Yellow` This one works it shows the proper code for "Yellow" = #ffff00
`Cyan`

but the other three are showing colors names that are different then yellow but if you click on their buttons to load the color pickers, they are showing "Yellow" = #ffff00

If I type these colors in a different color setting in `Application Preferences` such as `Graphic View - Entity Handle` color drop downs then these colors do work

Orange = #ffa500
Gray = #808080
Yellow" = #ffff00
Cyan = #00ffff

so I think it is specifically an issue with the color drop downs for the ones found in the `Application Preferences - Info Cursor` tab

closed 2026-11-07 by dxli