LibreCAD/LibreCAD #2143: Selection rectangle cursor now shows snaps

Questions2000, 2025-05-09:
BUG REPORT / FEATURE REQUEST?

### Expected Behavior
When using selection rectangle (activated by holding down LMB and drag to select items) would like to not have gray snaps show up.
Not sure if these is are on purpose / purpose of these gray snaps points when selection retangle is active?
Not knowing their purpose, to me they seem a little distracting.
I did not see this setting available, if it is not, suggesting to add a setting in "Application Preferences" to be able to turn these gray snaps "off"

### Observed Behavior
When you activate selection rectangle, as your cursor moves around near any drawn items, any snap settings you have "on" end up showing up in gray color.

### Similar Post
I had originally thought the dark gray snaps mentioned in this post above was specifically happening because of selection rectangle, but then realized that all my snaps are showing up dark gray as mentioned in the following post

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-415-gd06f27896.AppImage

sand1024, 2025-05-10:
@Questions2000 
Well, as for gray - I suppose it's related to #2144. 

As for "snaps" - I suspect you actually refer to entity handles (and there is separate colors settings for them). 
They are shown for the entity in selected state, regardless whether the entity was selected by the click on it or by other ways (like selection using rect area).  

I'll think about this more, yet so far I"m not sure it's good idea to hide these handles if more than one entity is selected..... Potentially, it's possible (at least with adding a setting for this) - yet the major question is whether such behavior will improve usability or not...

Questions2000, 2025-05-14:
No no, I mean snaps (not "entity handles")

I went back in my vault of old LC AppImages I have saved for testing purposes.
Best info I can give you is when I use this version
`LibreCAD-2.2.2_alpha1-271-ga6e5bfc7.AppImage`
Snaps = orange color as expected
and
There are no "snaps" that show up on the cursor when LMB click hold and drag to generate selection rectangles
Now of course for some reason I had a gap of AppImages missing in between so I cannot pin point exactly what version this occurs in, 

but when I try this version
`LibreCAD-2.2.2_alpha1-294-ge6f74e4e.AppImage`
Snaps = gray
and
Snap shows up when generating selection rectangles.

`More Detail About Selection Rectangle`
-The "snap" square shows up at the tip of arrow head mouse cursor the second the LMB is held down.
-If I click (this is the top left corner of the selection rectangle) and drag down and to the right, the selection rectangle is created and the mouse cursor arrow head is at the bottom right of the selection rectangle, the "snap" square remains at the tip of the mouse cursor.
-When let go of LMB and do not move mouse the "snap" square is still at cursor arrow until the mouse cursor is moved (the movement I believe refreshes the drawing pane which the refresh removes "snap" square from being visible.

Also I used fresh .config/LibreCAD folders so not a conflict with any older settings.

Questions2000, 2025-07-10:
Selection rectangle has no use for snaps.
Older versions of LC did not show them.
`LibreCAD-2.2.2_alpha1-271-ga6e5bfc7.AppImage`

Again this issue seems to have have happened at the same time as issue #2144 did.

Tried in newer version listed below and this still occurs.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-436-gc34b132b1.AppImage

Questions2000, 2025-08-02:
Just curious, is anyone else experiencing this?
Just wondering if anyone is willing to confirm this.
Kind of makes things difficult to use.
Thank You

Questions2000, 2025-09-09:
Tested in the following version and this still seems to be occurring.  I was wondering if the patch to fix this could be applied.
Thank You

### Operating System and LibreCAD Version
Version: LibreCAD-2.2.2_alpha1-457-gd1318274c.AppImage
Compiled on: Aug 29 2025
System: Debian GNU/Linux 12 (bookworm) / GTK based

Questions2000, 2025-09-29:
### Observed Behavior
Similar issue 2144 was fixed in the latest alpha 473 release, but unfortunately this issue still occurs.

The snap still shows up when create drag selection rectangle.
Now with issue 2144 being fixed, the snap is no longer gray, it is the default orange so it stands out more now.

When you create selection rectangles if mouse cursor is anywhere near any items, snap settings make the orange snap square shape go from the tip of the mouse cursor to those items.

This can be visually distracting.
Does anyone else experience this?

Thank You

### Operating System and LibreCAD Version
Version: LibreCAD-2.2.2_alpha1-473-g606035f42.AppImage
Compiled on: Sept 18 2025
System: Debian GNU/Linux 12 (bookworm) / GTK based

sand1024, 2025-11-08:
@Questions2000 hm... I still can't reproduce this. 

Is if possible for you to create a short screen recording to illustrated how exactly it looks on your system? Say, first on older version when they works fine and on the latest where they are broken? 
Probably such a screen recording may provide me with some insights...

Questions2000, 2025-12-18:
I do not use these settings personally but noticed this when testing out other issue where I needed to remove my `LibreCAD` folder from my `.config` folder which starts fresh launch of LC.

If user uses `Application Preferences - Snap - Snap Marker` setting `Snap Indicator Lines` this can be distracting to have these indicator lines moving around while trying to focus on selection rectangle.  This is even worse if user is not using the Snap toolbar item `Free Snap`.
With that setting off, when you try to make selections, the indicator lines jump all over screen, snapping to the nearest snap settings user has activated.

Wish there was some way to fix this?

Also, again, I was wondering, do any developers use Linux? or only Windows?  Do any developers who use Windows have access to testing LC AppImage / Linux in a VM?

Why would something like this only show up in Linux and Windows?

As I mentioned in my reply above
"So it is some time between versions 272-293 within a 10 day period."
does anyone know what could have happened in this 10 day period that caused this to occur?

Thank you to anyone who reads this