LibreCAD/LibreCAD #2144: Snap color shows does not appear to be working correctly

Questions2000, 2025-05-09:
### Expected Behavior
I apologize I might be overlooking something but it does not appear that "Snap" colors are working correctly on my machine.
I am expecting the snap color to match the color assigned in Application Preferences - Snap
Which by default `#ffc200` which is a light orange

### Observed Behavior
Snap color always appear as a dark gray.

The color in Preferences shows `#ffc200` which is a light orange

I turned off the following settings to be able to clearly see the snaps
`Snap - Hide cursor when snapping`
`Preview - Show operations preview explanatory entities`
`Info Cursor`

Drawing a line and snapping to end points for example show gray snaps
Dragging an existing lines end point shows the yellow guide line and the snap is yellow (not orange either)

(I also tried clearing out my entire settings file and this still occurs after I set everything up again)

### Similar Post
I had made this post originally thinking it was something with selection rectangle.
But the realized that all my snaps are showing up dark gray.
https://github.com/LibreCAD/LibreCAD/issues/2143

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-415-gd06f27896.AppImage.AppImage

sand1024, 2025-05-10:
@Questions2000 

well, that's a really weird behavior that you've described... I've never seen something similar. 

Could you please export your settings (from Application Preferences -> Paths -> Exchange->Export Settings to file) and attach resulting file for this ticket?

sand1024, 2025-05-10:
@Questions2000 
Btw, if by "snap" you actually refer to entity handles - please check your settings for colors for them. 
They are located in Application Prefs -> Graphic View -> Entity Handle

sand1024, 2025-08-03:
@Questions2000 I suppose that issue is fixed already, yet it not a part of the master branch yet and so it not part of the AppImage. 
Fix will be available in one of nearest updates.

Questions2000, 2025-09-09:
Tested in the following version and this still seems to be occurring.  I was wondering if the patch to fix this could be applied.
Thank You

### Operating System and LibreCAD Version
Version: LibreCAD-2.2.2_alpha1-457-gd1318274c.AppImage
Compiled on: Aug 29 2025
System: Debian GNU/Linux 12 (bookworm) / GTK based

Questions2000, 2025-09-29:
This works now, thank you for making this fix.

closed 2025-09-29 by Questions2000