LibreCAD/LibreCAD #2148: Feature request - Save (and load) layer state

Libarch, 2025-05-11:
(In the absence of xrefs) would it be easy to implement a feature allowing you to save the current state of the layer tree (visibility, layer attributes etc for each layer), and reload it later? Eg for architectural plans you could save an appropriate layer state for Proposed Layout, Reflected Ceiling Layout, Existing Layout and so on.

sand1024, 2025-05-12:
@Libarch 
Yes, something similar is planned and it should be included into one of the future update (may be not the nearest, yet still). 

This will be hardly portable solution  (as there is no support for layer state in DXF, AFAIK) - but it is achievable at least with external config file (like Layer State (LAS) File in AutoCAD)

Libarch, 2025-05-12:
Thank you. I've never used AutoCAD, but [this](https://help.autodesk.com/view/ACD/2022/ENU/?guid=GUID-5312A8BD-DD94-47D6-B1BA-5E0AF5E0CED8) is more or less exactly what I had in mind. I hope it's achievable.