LibreCAD/LibreCAD #2150: Categories toolbar - Change circle tools label to Circles

Questions2000, 2025-05-14:
### Observed Behavior
In the `Categories` toolbar that by default is vertical on LEFT side of screen.
If you hover mouse over the "Circles" icon it shows tag label that says `Polygons`
In the LEFT dock toolbar widget it contains tabs for each tool group and the tab for "Circles" section is called `Circles`
Suggesting to change the `Polygons` hover tag label that shows in "Categories" toolbar to `Circles`

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-415-gd06f27896.AppImage

sand1024, 2025-05-14:
@Questions2000 
oh, it's just a nasty copy/paste bug, thank you for reporting. For sure it should be changed...

Questions2000, 2025-07-10:
Still occurs

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-436-gc34b132b1.AppImage

closed 2026-09-27 by sand1024