LibreCAD/LibreCAD #2152: Polygon tools create invisible polygons in drawing pane

Questions2000, 2025-05-14:
### Observed Behavior
When using any of the `Polygon` tools within the `Polygon` category
Polygon CenCor, CenTan, CorCor, TanTan
When you draw any polygons using these tools, the drawing preview of the shape shows up but when you click to complete the polygons they do not appear to draw the shape.
Appear being the keyword, because at first I thought the items were just not drawing, but after you create a polygon shape and it does not appear, if you hover your mouse around the area where the shape would show up, the hover cursor over item highlight works on these invisible item, making them visible until you move your mouse cursor away from the area.
If the "Polyline" check box in Tool Option toolbar is NOT checked then the highlight hover will show a bunch of line pieces, that make up the polygon shape you create that is invisible.

Easiest way to test this is to just use one of  4 Polygon tools to create shapes in a blank drawing pane.

I also tried for example the TanTan with two circles already drawn on screen.

Just wondering if someone could verify this is happening on their machine.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-415-gd06f27896.AppImage.AppIma

Questions2000, 2025-11-13:
Thank you for confirming

I recently tested this out again and this still occurs.
However in my recent tests it appears that parts of this issue now are different.
So this was probably improved a little by a different update.
This now appears to generate more of a refresh based issue instead.
I have updated my original thread post above if any developers or users are interested.

Thank You

sand1024, 2026-09-27:
can't reproduce it.

closed 2026-09-27 by sand1024

Questions2000, 2026-10-10:
Tested and works great now.
Thank you for making this fix