LibreCAD/LibreCAD #2153: Zoom panning on alpha builds cancels current command

ksio89, 2025-05-15:
**Observed Behaviour**

Zoom panning tool on alpha builds cancels current command

**Intended Behaviour**

zoom panning tool on alpha builds does not cancel current command 

I don't know if it's a bug or just undesired behaviour, but on alpha builds the zoom panning tool, accessed by pressing mouse scrollwheel button, cancels the current command, which is extremely annoying and did not happen on latest release candidate build. I looked everywhere on Application Preferences for an option that disables this behaviour, but could not find anything. 

**Operating System and LibreCAD Version**

Version: 2.2.2.3-alpha
Compiler: Microsoft Visual C++
Compiled on: May  9 2025
Qt Version: 6.9.0
Boost Version: 1.87.0
System: Windows 11 Version 24H2

sand1024, 2025-05-17:
@ksio89 
Yes, this is intentional yet works rather like a bug, so this be changed in next builds

sand1024, 2025-06-06:
@ksio89 
Hm... it's hard to say the particular build where it was introduced. 

Do you rely on pre-made image or build from the sources, btw? 

Actually, my local sources where this is already fixes are far from be ready to commit, as at the moment I'm working on quite a big feature and the codebase is a bit broken. 

I planned that I'll be able to commit something solid in about nearest couple of weeks and there this issue will be fixed among others. 

However, I'll take a look to that fix and probably I'll be able to commit change for this only one of these days.

closed 2026-09-27 by sand1024