LibreCAD/LibreCAD #2154: spline2 not accessable via "load command file"

Jim-2249, 2025-05-18:
I have not been able to draw a spline2 via the "load command line". I tried both as a single line with semicolons as well as separate line w/o semis.   I also tried a number of other undocumented file formats without success. Semi-separated version shown for brevity.
The following will draw two lines:
`li; 0,0; 1,1; 2,0; k;`
The following, using either "stp" or "spline2" does not generate anything:
`stp; 0,0; 1,1; 2,0; k;`
The following is echoed in the command line box after the "stp" command file is loaded:
 ```
stp
Command: stp (spline2)
0,0
1,1
2,0
k
Command: k (kill)
Command: k (kill)
Command: k (kill)
Spacebar: temporarily set snapping mode to free snapping
```
Note: "snapping" line appears only if a semi follows the "k". (Why the multiple "Command: k (kill)"? There was only one "k")

After executing any of those command files, a red point "Relative zero" is displayed at 2,0
A file containing only "spline2" or "stp" will start a spline2 that I can manually select points.

**Am I doing something wrong here?** Using a file to draw spline through numerous points would be useful.

This issue was mentioned on the [forum Oct 12 2020](https://forum.librecad.org/Spline-through-points-from-command-file-td5719617.html) There were no replies.

Version: v2.2.1.1
Compiler: GNU GCC 12.2.0
Compiled on: Feb 18 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009
**Note: "About" gives wrong windows version.** System Windows specs:
Edition: Windows 11 Home, Version: 24H2, Installed on: 2/‎21/‎2025, OS build: 26100.4061
Experience: Windows Feature Experience Pack 1000.26100.84.0

Jim-2249, 2025-05-19:
I found a work-around. If I leave off the "k", it will add the spline, but it's not terminated. It the end is still attached to the cursor. I need to terminate the spline manually with a right-click.

dxli, 2025-05-26:
@Jim-2249 

K or kill should terminate the command and add the spline here.

dxli, 2025-06-03:
@Jim-2249 

It appears that the commands with the scope of an active spline through points, i.e. the commands to control an action behavior, are not working at all.

I feel this is a major regression, and should be fixed at high priorities.

Thanks a lot for bringing this up.