LibreCAD/LibreCAD #2160: Circle tangent to 3 Cirles fails

hhassey, 2025-05-26:
While trying to create a fourth circle tangent to other 3 circles, LibreCAD fails to draw the circle in this particular arrangement. 

[image]

It will not let you select the third circle no matter what order you choose them .

dxli, 2025-05-27:
@hhassey 

Could you upload a sample dxf with this issue? I cannot reproduce this issue

[image]

dxli, 2025-05-30:
The root cause: failure of the Algebraic algorithm as described here https://en.wikipedia.org/wiki/Problem_of_Apollonius
The fix in [3c4196e](https://github.com/LibreCAD/LibreCAD/commit/3c4196e185308e1ff1ec6e107b05cfb557a3df90): when the algebraic algorithm fails, falls back to a generic center finding algorithm by the paths of shared tangential circles of two given circles.

closed 2025-08-10 by dxli