LibreCAD/LibreCAD #2162: ARM MacOS builds?

silasb, 2025-05-27:
I'd like to use ttf2lff but after installing freetype and trying relink freetype to the correct location things didn't work.  That is when I noticed that the binaries are built for X86.  Trying to get a freetype X86 build on my M1 is probably not worth the effort.

If we have ARM MacOS builds this should eliminate some of these issues.

dxli, 2025-05-27:
@silasb 

probably, we need to build ibfreetype from source for arm support.

please have a look at our building script, and we need arm support for mac anyway

[building workflow
](https://github.com/LibreCAD/LibreCAD/blob/master/.github/workflows/build-all.yml)

silasb, 2025-05-28:
I can help.  When I get back from vacation I'll setup a couple different VMs for testing.

dxli, 2026-01-06:
it appears that mac Applie Silicon will be more restrictive to unsigned apps.

To get the app signed requires an Apple Developer ID, which costs $99 per year.