LibreCAD/LibreCAD #2168: Snap middle manual stopped working

gerardbm, 2025-06-05:
I'm using the following LibreCAD version:

```
Version: 2.2.2.3-alpha
Compiler: GNU GCC 13.3.0
Compiled on: Jun  5 2025
Qt Version: 6.4.2
Boost Version: 1.83.0
System: Debian GNU/Linux 12 (bookworm)
```

Expected behaviour: you get the line (for example) to start drawing. Then you activate the Snap Middle Point, and you can now select 2 points (A and B), because it will insert the start of the line in the middle point of this 2 points A and B. Now, Snap Middle Point is automatically disabled. That worked!

What really happens? You get the line (for example) to start drawing. Then you activate the Snap Middle Point, and now the line is disabled. You can't select 2 points (A and B), so you can't insert a line between two points. Now, Snap Middle Point remains enabled without working...

Check this video: https://www.youtube.com/watch?v=4pGMmmIxgbI

gerardbm, 2025-06-07:
Hi, @sand1024. Yes, of course! [https://www.youtube.com/watch?v=pC9ZDJIKZhU](https://www.youtube.com/watch?v=pC9ZDJIKZhU)

During the first minute, it's an older LibreCAD version I can use in my Debian 9. That worked.
The second minute of the video is the last LibreCAD version in Debian 12 with virtual box.

This snap tool is a little different from the others, because it must be disabled automatically every time you have used it successfully: that means, after entering the second point.

BTW, I'm seeing that this snap tool worked fine, HOWEVER, not always was good. After some uses, it stopped working as well. It ended using the 100% of the distance instead of the 50%. So if you draw a line and then you want to end it between two points, it uses the second point instead of the half distance. The same happens with other actions like copying from A to B, etc. Without closing LibreCAD, it started to work again... So I think it needs a good revision.

You can check a good behaviour in QCAD CE and Allplan. It's a super useful snap when it works fine.

dxli, 2025-06-07:
@sand1024 

Yes, some logic in event handler and action handler need to be reviewed. The current state is arrived by constant patching of quick fixes.

ErtugrulEge, 2025-06-13:
I cab also confirm the behaviour