LibreCAD/LibreCAD #2180: Polygon tools drawing invisible polygons

ErtugrulEge, 2025-06-10:
Greetings everyone!

Steps to reproduce:
1.Select any polygon tool and try to draw a polygon you can't see anything at first
2.Exit from command
3.Hover near where you try to draw polygon you will see the edges of invisible polygon

Version: 2.2.2.3-alpha<br/>Compiler: Microsoft Visual C++<br/>Compiled on: Jun  5 2025<br/>Qt Version: 6.9.1<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

[image]

gerardbm, 2025-06-10:
This issue is probably a duplicate of this one: [2152](https://github.com/LibreCAD/LibreCAD/issues/2152)

ErtugrulEge, 2025-06-10:
> This issue is probably a duplicate of this one: [2152](https://github.com/LibreCAD/LibreCAD/issues/2152)

Sorry, I forgot to serarch.