LibreCAD/LibreCAD #2182: Dimension tool creates incorrect dimension

ErtugrulEge, 2025-06-11:
Greetings everyone!

Steps to reproduce:
1.Change unit to centimeter from `Current Drawing Preferences`
2.Be sure Length factor is 1 
3.Draw a line with entering 100 to create a 100cm line
4.Make a dimension, you will see dimension text will be 10 instead of 100

Version: 2.2.2.3-alpha<br/>Compiler: Microsoft Visual C++<br/>Compiled on: Jun  5 2025<br/>Qt Version: 6.9.1<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

[image]

dxli, 2025-06-13:
@ErtugrulEge 

I cannot reproduce this.

[image]

ErtugrulEge, 2025-06-13:
You are right. I tried with another computer and everyting worked fine. I tried to reset all my settings but I couldn't find a way. I tried uninstall and delete AppData  but they didn't work.

https://github.com/LibreCAD/LibreCAD/issues/2191

closed 2025-06-18 by dxli