LibreCAD/LibreCAD #2189: Grid is shown inspite of  the Application Preferences grid checkbox is off

ErtugrulEge, 2025-06-13:
Greetings everyone!

I don't want to see grid so I changed gird setting from Application Preferences but I can still see the grid.When I change application
preferences Shouldn't I see the grid?

Version: 2.2.2.3-alpha<br/>Compiler: Microsoft Visual C++<br/>Compiled on: Jun  5 2025<br/>Qt Version: 6.9.1<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

[image]

sand1024, 2025-06-13:
@ErtugrulEge 
Well, I think that depends on what you're going to achieve. 

The setting you've illustrated is used to control grid visibility for newly created drawings. 

If you need to disable the grid, you can either rely on the button in toolbar, or use a different settings as it shown by the image  below. 

[image]

ErtugrulEge, 2025-06-13:
Thank you for the tip.Those settings work but I want to repeat it again to avoid misunderstanding.
1.I turned off grid checkbox from application preferences 
2.I close LibreCAD
3.I opened LibreCAD, Now, a new drawing window opened. Should I see grid dots or not?

closed 2025-06-18 by dxli