LibreCAD/LibreCAD #2200: `Offset`  works incorrectly

ErtugrulEge, 2025-06-16:
Greetings everyone!

Steps to reproduce:
1.Create a polyline 
2.Press `Offset`  button and  set a distance and press polyline

Version: 2.2.2.3-alpha<br/>Compiler: Microsoft Visual C++<br/>Compiled on: Jun  5 2025<br/>Qt Version: 6.9.1<br/>Boost Version: 1.87.0
System: Windows 10 Version 22H2

[offset_test.zip](https://github.com/user-attachments/files/20758247/offset_test.zip)

sand1024, 2025-06-16:
@ErtugrulEge 
Well, that's not an in issues actually, it's rather historical implementation of the Offset command.  
The offset if mostly used for atomic entities, for making proper offset of polylines - it's necessary to use "Create Equidistant Polyline" action.  The "Offset" applied to polyline will make offset of individual segments (and so, there will be segments after intersections and so on).

ErtugrulEge, 2025-06-16:
I do  the same things with  2.2.1.1 and  offset works as expected.

Version: v2.2.1.1
Compiler: GNU GCC 12.2.0
Compiled on: Feb 18 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009

[image]

sand1024, 2025-06-18:
@ErtugrulEge 
Hm... ok, that's strange, I'll take a look on this once more