LibreCAD/LibreCAD #2206: Hatch issue: clicking 'enable preview' causes a segmentation fault

gerardbm, 2025-06-18:
LibreCAD version:

Version: 2.2.2.3-alpha<br/>Compiler: GNU GCC 13.3.0<br/>Compiled on: Jun  5 2025<br/>Qt Version: 6.4.2<br/>Boost Version: 1.83.0
System: Debian GNU/Linux 12 (bookworm)

I'm executing the file `LibreCAD-2.2.2_alpha1-422-g9a1fb2aef.AppImage` under Debian 12.

There's an issue in the hatch tool: when you click to enable preview, it causes a segmentation fault.

Steps to reproduce:

1. Draw a figure, like a rectangle or four lines closed.
2. Select them
3. Select the hatch tool
4. Click "enable preview"

PS. BTW, only the "concrete" texture was appearing in the select box.

dxli, 2025-06-20:
This appears to be an AppImage issue.

ksio89, 2025-06-23:
@ErtugrulEge off-topic, but how did you get the classic light theme on build 2.2.2.3 alpha?

ErtugrulEge, 2025-06-24:
> [@ErtugrulEge](https://github.com/ErtugrulEge) off-topic, but how did you get the classic light theme on build 2.2.2.3 alpha?

Options->Widget options

[image]

dxli, 2025-11-17:
cannot reproduce. closing

closed 2025-11-17 by dxli