LibreCAD/LibreCAD #2210: Freeze on snap (in RS_EntityContainer)

sand1024, 2025-06-22:
@dxli 
Hi!
I've found quite a strange issue that leads to freeze when making a snap (on master branch).   The freeze is reproduceable on [LibreCAD-2.2.2_alpha1-429-gbb314fad1-win64.exe](https://github.com/LibreCAD/LibreCAD/releases/download/2.2.2_alpha-latest/LibreCAD-2.2.2_alpha1-429-gbb314fad1-win64.exe), so it's not related to my local modifications. 

The freeze is due to infinite cycle. That infinite cycle occurs in RS_EntityContainer::getNearestIntersection, here:  

[image]

I suspect it's related to your recent changes in RS_EntityContainer, and the logic for the cycle termination is broken somewhere. 

Actually, that infinite cycle is hard to reproduce reliably - it seems to be floating that occurs just from time to time  (at least, I've got it several times).  
Potentially, active moving of the mouse may help.  I've got one on several different drawings, so it seems that it's not related to specific drawing data.  

However, it seems that this feeze is reproduceable on the attached drawing - just move mouse around. 

[lc_freeze.zip](https://github.com/user-attachments/files/20852045/lc_freeze.zip)

Snap modes that are enabled are as following:

[image]

dxli, 2025-06-22:
@sand1024 

I haven't been able to reproduce the issue. However, there's a design issue with RS_EntityContainer::first() and next():

the traversing is controlled by member variables "subcontainer" and "entIdx", it means any mult-threading traversing would trigger undefined behaviors. I feel we should redesign the first()/next() mechanism based on clean graph DFS traversing, and the traversing should not modify the target container.

My previous change was triggered by troubleshooting of #2179. Incorrect RS_EntityContainer copy ctor/assigner led to corrupted data models. The hatch corruption was only a symptom.

dxli, 2025-06-23:
@sand1024 

I will modify the first/next/previous functions to be safer.

dxli, 2025-07-13:
@sand1024 

I have merged the new container traverser. For most use cases, the traverser collects all entities into a std::vector, and the code   simple and more maintainable than before.

For generator type traversing (one entity a step), the code is not thoroughly tested, and we will fix any bug come across.

I hope the traverser concept should have fixed this issue.

closed 2025-08-10 by dxli