LibreCAD/LibreCAD #2211: Dimension lines has gaps close to the arrows.

pa36651, 2025-06-28:
With the latest release there is a gap close to the arrows. I cant find any setting for this, assuming it is a bug.

[image]

dxli, 2025-06-29:
Are you running the latest development or 2.2.1?

Could you provide a minimum DXF file with this issue?

pa36651, 2025-06-29:
Hi,
Tank you for quick response.
I am using 2.2.1.
Please find a dxf attached.

⁣Mvh
Petter Almhagen 
0707812510
Windbreaker AB ​

dxli, 2025-06-30:
I remember this regression was fixed in the 2.2.1 branch.

Please help us out by testing the latest 2.2.1 builds:

https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.1.1_rc-latest

We have accumulated quite a few fixes to justify a 2.2.1.2 bug release.