LibreCAD/LibreCAD #2214: Window is not maximizing, doesn't fit/resize

eugenialoli, 2025-06-29:
I downloaded the latest available .appimage file from April, and the app doesn't maximize (the maximize button is missing on the window manager on this app), and also, it doesn't fit on the 1366x768 resolution of my laptop (many laptops still use that res). The window doesn't resize when I try to resize it! I'm using Linux Mint with X11.

[image]

dxli, 2025-06-30:
@eugenialoli 

Can you hide the "Pen Widget" from Menu: Widgets > "Pen Widget", or "Pen Wizard"?

The direct cause of the issue is due to the tool dock area takes up more space than the actual screen height.

If this trick works for you, please let us know, and we may consider to keep those widgets hidden for new installations.

eugenialoli, 2025-06-30:
Thank you for the reply. No, this didn't fix the overly long window. Even if it did, it still wouldn't fix the fact that the app window doesn't resize or enable the maximize button on the window manager...

dxli, 2025-07-03:
@sand1024 

in 2.2.2-alpha, we probably want to add a scroll area for the right dock area.

sand1024, 2025-07-03:
> [@sand1024](https://github.com/sand1024)
> 
> I disabled showing the "Pen Wizard" by default. by [bffb961](https://github.com/LibreCAD/LibreCAD/commit/bffb961c54be21fdcdfd45be4db43f92558e8832)
> 
> I tested it works for me on a small screen linux virtual machine, meaning the main window is visible starting from scratch.
> 
> [@eugenialoli](https://github.com/eugenialoli)
> 
> To work around by disabling "Pen Wizard":
> 
> 1. start the app;
> 2. Main menu: Widgets->Dock Widgets->Pen Wizard (to hide the Pen Wizard);
> 3. Un-maximize the app window, and maximize it again to show the whole app window.

@dxli 
Yes, that's quite reasonable. 

Basically, it might be practical to hide all widgets except the CMD by default. 
That will also force the user to set up the workspace according to their needs (ideally, a "fist run" setup wizard should be expanded to provide more details about UI customization... 

> in 2.2.2-alpha, we probably want to add a scroll area for the right dock area.

Hmm... technically it could be (yet for sure a setting in UI is needed)—but I'm almost sure that it might lead to usability effects... Scrolling for the widget — it's too slow, as for me, plus there will be double scrolling (scrool for the widgets list + own scroll area of specfic widgets).  I think that current docking support is more than enough for proper space organizing. 

For sure, the initial layout of the screen may be improved, yet I'm not sure I understand why we should bother there more than some minimal efforts. And why  the user can't organize the workspace via drag-n-drop and docking if something does not fit—even on such a small display...