LibreCAD/LibreCAD #2218: Negative fractional numbers not displayed properly

vtkovacs, 2025-07-09:
In "Current Drawing Preferences" if the Format of Units is Fractional negative numbers are is displayed incorrectly - example:
 -21 4294965949/32
Screen shot

[image]

 Version: v2.2.1.1
Compiler: GNU GCC 9.4.0
Compiled on: Feb 18 2025
Qt Version: 5.15.2
Boost Version: 1.71.0
System: Fedora Linux 41 (Workstation Edition)

This version has the same issue:

Version: 2.2.2_alpha1-436-gc34b132b1<br/>Compiler: GNU GCC 14.3.1<br/>Compiled on: Jul  9 2025<br/>Qt Version: 6.8.3<br/>Boost Version: 1.83.0
System: Fedora Linux 41 (Workstation Edition)

dxli, 2025-07-10:
The root cause: RS_Units::formatFractional() fails to consider negative values.

Fixed in 2.2.1 and master.

vtkovacs, 2025-07-10:
Thanks! Downloaded v2.2.1.1-19-g3ff2e1515 and it works perfectly.

closed 2025-07-10 by vtkovacs