LibreCAD/LibreCAD #2225: Dialog Windows - Tab cycling hotkeys

Questions2000, 2025-07-10:
### Expected Behavior
To be able to use `Ctrl+Tab` / `Shift+Ctrl+Tab` hotkeys to cycle through tabs in any dialog windows that contain them.
These hotkeys should work with or without focus being in tab bar.

### Observed Behavior
`Ctrl+Tab` / `Shift+Ctrl+Tab` hotkeys do not always work as expected in dialog windows that contain tabs.
The main ones I have found are `Application Preferences` & `Drawing Preferences` windows.

I think the underlying issue causing this is actually the following hotkeys instead
`Tab` = forward focus
`Shift+Tab` = backwards focus
In these windows, it appears that the only way that these `Ctrl+Tab` / `Shift+Ctrl+Tab` hotkeys work is when tabs get focus.

### Steps To Reproduce
`Application Preferences` & `Drawing Preferences`
If you open up one of these window and try `Ctrl+Tab` to advance to next tab it does not work.
If you hit your `Tab` key then you will notice that the first tab gets highlighted/receives focus.
Once this is done, then `Ctrl+Tab` / `Shift+Ctrl+Tab` work, as well as `Left & Right Arrows`, allow you to cycle through the tabs.

`Layers Tree Settings/Options Dialog`
In this dialog window `Ctrl+Tab / Shift+Ctrl+Tab' keys work as expected.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-436-gc34b132b1.AppImage

sand1024, 2025-07-12:
@Questions2000 
Isn't it a duplicate for LibreCAD#2226?

Questions2000, 2025-07-18:
@sand1024 
No, that is just the fun of having "Tab" key being similar to "Tab" bar.
And there is also overlap between the two issues as well.

2225 = This post is about cycling through the Tab bar at top of any dialogs that contain tabs, using common hotkeys `Ctrl+Tab`

2226 = Other post is about how the `Tab` key cycles through the settings themselves
The overlap I mentioned is that while cycling through the actual settings themselves, the `Tab` key also eventually cycles focus into the `Tab Bar` of these dialogs

Hope this makes sense.

Again I am just mentioning things that I notice, obviously up to you guys what you feel is worth updating.