LibreCAD/LibreCAD #2228: Importing DWG keeps the mouse cursor with the loading animation

NicoCaldo, 2025-07-11:
The .dwg import works fine (annoying the continue pop-up window saying it is not complete, but anyway).

[image]

The only issue is that, after clicking on the drawing and importing it, the mouse cursor keeps getting the loading icon of Windows and it does not switch back to the cursor

dxli, 2025-07-13:
@NicoCaldo 

Could you elaborate a little?

Version: 

Screenshots:

NicoCaldo, 2025-07-14:
Hi @dxli 
Unfortunately, I can't take screenshots with the cursor, so here is a video.
Version is
Version: v2.2.1.1-20-g63519937a
Compiler: GNU GCC 12.2.0
Compiled on: Jul 10 2025
Qt Version: 5.15.2
Boost Version: 1.87.0
System: Windows 10 Version 2009

https://github.com/user-attachments/assets/166f5130-b312-409e-a2ac-c51cec8f3d6d

dxli, 2025-07-14:
@NicoCaldo 

From the video, it appears to me the imported entities are not valid, so causing some internal processing errors, hence leaving the loading cursor as a symptom.

Unfortunately, we currently don't have a developer working on DWG support. The best approach is to use some external DWG converter to generate the DXF format, before loading by LibreCAD.

sand1024, 2025-07-17:
@dxli 
What do you think, probably it's better to drop DWG support at all? 

Support of DWG is quite limited, and I really doubt we could support various versions of AutoCAD there (especially given that DWG is not properly described and is proprietary that might be changed from version to version).   

Thus I think it might be just misleading for the user that tries to import DWG—as it is hard to identify what is actually wrong with DWG or degrade gracefully... 

If DWG is dropped, then DXF will be the primary format (probably together with its binary form). 
And if somebody needs to work with DWG for interoperability—some external converter should be used indeed. 

Also, I'm really not sure that JWW format support is in good shape—it was not updated for a long time (while I can see that new versions of JWW are being released (https://www.jwcad.net/versioninfo.htm)... 

So it's interesting if someone opens JWW files at all using LibreCAD..