LibreCAD/LibreCAD #2230: Connect two points in a "L" shape

TheDevick, 2025-07-12:
Hi everyone!

I'm currently working on a project where I need to connect multiple points using L-shaped lines, like in the images below:

[image]
[image]

Right now, I'm using the "2-point line" tool with Snap to Endpoints enabled, and entering the coordinates manually in the command line. I check where the second point is and then type the intermediate coordinates to create the L-shape manually.

Is there a more efficient way to do this?
Ideally, I'd like to just select the first and second point, and have the program automatically draw two lines (horizontal and vertical) to connect them in an L-shape.

Thanks in advance 🙏

dxli, 2025-07-13:
@TheDevick 

You can probably try Tools > Snake (tools). Probably, command line coordinate input helps.

lordofbikes, 2025-07-13:
This is a good question for our forum, where experienced users may have had the same task.
https://forum.librecad.org/

I'd use the parallel tool for this.