LibreCAD/LibreCAD #2233: Show release number in Alpha About dialog

Questions2000, 2025-07-19:
Each alpha release filename looks similar to the following
`LibreCAD-2.2.2_alpha1-439-gc34b132b1.AppImage`

Not sure what the exact name for this is, but I am referring to it as "release number"
Release Number = 439 (from example above)
I find this number very helpful to quickly know what release I am testing.

Suggesting maybe can add this release number somewhere in the Alpha release `About` dialog.

Not as valuable to me but maybe others would also find value in showing the "commit code" as well, but that is up developers if feel it is needed?
Commit Code = gc34b132b1 (from example above)

Thank You

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-439-gc34b132b1.AppImage

sand1024, 2025-07-21:
@Questions2000 yes, that's reasonable 

@dxli could you please take a look at this? If the version of LC corresponds to the build number, and so there will be more consistent version numbering— a new version check notification will also work for development versions.  

I'm referring there to the **LC_VERSION** variable that is set somewhere in build scripts (in CMake - as 

`add_compile_definitions(LC_VERSION=2.2.2.3-alpha)`

and that is later used as 

` QCoreApplication::setApplicationVersion(XSTR(LC_VERSION));`

It will be ideal if **LC_VERSION** correspond to the tag version on GitHub, so the list of tags received via 

> https://api.github.com/repos/LibreCAD/LibreCAD/releases

may be properly matched. 

I'm not sure I understand how the build number is created on GitHub, so I'm afraid I can't modify this by myself.

Questions2000, 2026-08-29:
Probably will not be added, decided to close.