LibreCAD/LibreCAD #2235: Widget docks refresh when closed and reopened

Questions2000, 2025-07-19:
POSSIBLE BUG REPORT

### Expected Behavior
To have the widget docks remember previous layout when close and reopened

### Observed Behavior
(Probably happens in other docks as well)
If you edit/rearrange left or right docks (for example close some tabs, change the layout of how many widgets shown at same time, etc.), if you hit "Left" or "Right" dock buttons in the `Dock Area` toolbar to turn off the left/right dock, then hit the same button to toggle that dock back "on". the previous layout you setup gets refreshed, which means the user loses the desired layout they setup.

User might just be hiding docks briefly to gain more drawing pane screen space, but imagine would still want the previous widget layout they setup.

Developer can add a way to "refresh" the docks to default state if want to (could be useful), but I feel the default should save the desired layout setup by the user, when these docks or shown/hidden.

### Operating System and LibreCAD Version
Linux / Debian 12 based distro / GTK based Desktop Environment
LibreCAD-2.2.2_alpha1-439-gc34b132b1.AppImage

Questions2000, 2026-10-01:
I mentioned this a while back.
This probably should be looked at if anyone is interested, strange behavior.
I tested this again and it still occurs.

Will just mention a couple observations again

### Observed Behavior 1
-If you close (remove) some tabs
Example:
Say you close the `Layers` tab, because the `Layers Tree` widget is meant to replace it.
If you hide the RIGHT dock and open it again, the `Layers` tab returns.

### Observed Behavior 2
-Open RIGHT dock
-Horizontal tabs
-Divide the RIGHT dock into 2 sections
-Have each section contain multiple widget tabs
-In both sections have the first tab at the left of tab bars active
-Then keep closing and opening the RIGHT dock
-Notice that the active widget tabs change in each section
Keeps toggling from the 1st tab - 2nd tab - 1st tab - etc.

-----------------------------------

### Scenario
-Say user takes the time to setup the RIGHT dock exactly the way they want it displayed.
-That user may regularly need to temporarily hide the RIGHT dock at times, to provide more screen space for the drawing pane.
-Every time they do this, the RIGHT dock randomly resets itself, changes the active widgets user was previously using before dock was closed, etc.

Thank you to anyone who reads this

### Operating System and LibreCAD Version
LibreCAD-2.2.2_alpha1-710-g55bd61f34-x86_64.AppImage
Compiler: GNU GCC 13.3.0
Compiled on: Sep 28 2026
Qt Version: 6.4.2
Boost Version: 1.83.0
System: Debian GNU/Linux 13 (trixie)

Questions2000, 2026-10-18:
This seems like it has been fixed now.
Has this been worked on?

Seems to have happened between the following 2 versions
LibreCAD-2.2.2_alpha1-724-g30751cffb-x86_64.AppImage (issue occurs)
LibreCAD-2.2.2_alpha1-735-g3919b0e11-x86_64.AppImage (seems fixed)

If so thank you to whoever worked on this update.

Or maybe this was fixed with a fix for a different issue?

I will close/or developer can close once confirmed.

Questions2000, 2026-10-23:
@dxli 
@sand1024 
Would either of you be able to confirm if this has been worked on?
Not sure if I should close or not.
Thank You

sand1024, 2026-10-24:
@Questions2000 
this is fixed locally yet not released to public yet. Should be retested later.